RPi-Distro/raspi-config #3: Can't change memory split on a Raspberry Pi 2

voltagex, 2015-12-23:
[image]

Is there a way to get verbose output?

XECDesign, 2015-12-23:
No verbose output, but could you provide the output of the following commands:
```
ls -la /boot
mount
cat /boot/config.txt
```

voltagex, 2015-12-23:
Well that's a hot mess. I'm not sure you'd want to support my config.

```
voltagex@kiwi:~$ ls -la /boot
total 32656
drwxr-xr-x  4 root root    4096 Dec 20 18:09 .
drwxr-xr-x 22 root root 1978368 Dec 20 18:04 ..
-rw-r--r--  1 root root   10212 Dec 20 18:06 bcm2708-rpi-b.dtb
-rw-r--r--  1 root root   10491 Dec 20 18:06 bcm2708-rpi-b-plus.dtb
-rw-r--r--  1 root root   10242 Dec 20 18:06 bcm2708-rpi-cm.dtb
-rw-r--r--  1 root root   11479 Dec 20 18:06 bcm2709-rpi-2-b.dtb
-rw-r--r--  1 root root   17900 Dec 20 18:06 bootcode.bin
-rw-r--r--  1 root root  118446 Feb  3  2015 config-3.18.0-trunk-rpi2
drwxr-xr-x  2 root root   16384 Jan  1  1970 firmware
-rw-r--r--  1 root root      41 Dec 20 18:09 .firmware_revision
-rw-r--r--  1 root root    2472 Dec 20 18:06 fixup_cd.dat
-rw-r--r--  1 root root    6450 Dec 20 18:06 fixup.dat
-rw-r--r--  1 root root    9692 Dec 20 18:06 fixup_db.dat
-rw-r--r--  1 root root    9696 Dec 20 18:06 fixup_x.dat
-rw-r--r--  1 root root 5572712 Sep 28 14:05 initrd.img-3.18.0-trunk-rpi2
-rw-r--r--  1 root root 4024732 Dec 20 18:06 kernel7.img
-rw-r--r--  1 root root 3946040 Dec 20 18:06 kernel.img
drwxr-xr-x  2 root root    4096 Dec 20 18:06 overlays
-rw-r--r--  1 root root  605272 Dec 20 18:06 start_cd.elf
-rw-r--r--  1 root root 4878728 Dec 20 18:06 start_db.elf
-rw-r--r--  1 root root 2731832 Dec 20 18:06 start.elf
-rw-r--r--  1 root root 3831848 Dec 20 18:06 start_x.elf
-rw-r--r--  1 root root 2292595 Feb  3  2015 System.map-3.18.0-trunk-rpi2
-rw-r--r--  1 root root 3304216 Feb  3  2015 vmlinuz-3.18.0-trunk-rpi2
voltagex@kiwi:~$ mount
/dev/mmcblk0p2 on / type ext4 (rw,relatime,discard,errors=remount-ro,data=ordered)
devtmpfs on /dev type devtmpfs (rw,relatime,size=468844k,nr_inodes=117211,mode=755)
sysfs on /sys type sysfs (rw,nosuid,nodev,noexec,relatime)
proc on /proc type proc (rw,nosuid,nodev,noexec,relatime)
securityfs on /sys/kernel/security type securityfs (rw,nosuid,nodev,noexec,relatime)
tmpfs on /dev/shm type tmpfs (rw,nosuid,nodev)
devpts on /dev/pts type devpts (rw,nosuid,noexec,relatime,gid=5,mode=620,ptmxmode=000)
tmpfs on /run type tmpfs (rw,nosuid,nodev,mode=755)
tmpfs on /run/lock type tmpfs (rw,nosuid,nodev,noexec,relatime,size=5120k)
tmpfs on /sys/fs/cgroup type tmpfs (ro,nosuid,nodev,noexec,mode=755)
cgroup on /sys/fs/cgroup/systemd type cgroup (rw,nosuid,nodev,noexec,relatime,xattr,release_agent=/lib/systemd/systemd-cgroups-agent,name=systemd)
cgroup on /sys/fs/cgroup/cpuset type cgroup (rw,nosuid,nodev,noexec,relatime,cpuset)
cgroup on /sys/fs/cgroup/cpu,cpuacct type cgroup (rw,nosuid,nodev,noexec,relatime,cpu,cpuacct)
cgroup on /sys/fs/cgroup/devices type cgroup (rw,nosuid,nodev,noexec,relatime,devices)
cgroup on /sys/fs/cgroup/freezer type cgroup (rw,nosuid,nodev,noexec,relatime,freezer)
cgroup on /sys/fs/cgroup/net_cls,net_prio type cgroup (rw,nosuid,nodev,noexec,relatime,net_cls,net_prio)
cgroup on /sys/fs/cgroup/blkio type cgroup (rw,nosuid,nodev,noexec,relatime,blkio)
cgroup on /sys/fs/cgroup/perf_event type cgroup (rw,nosuid,nodev,noexec,relatime,perf_event)
systemd-1 on /proc/sys/fs/binfmt_misc type autofs (rw,relatime,fd=23,pgrp=1,timeout=300,minproto=5,maxproto=5,direct)
mqueue on /dev/mqueue type mqueue (rw,relatime)
debugfs on /sys/kernel/debug type debugfs (rw,relatime)
configfs on /sys/kernel/config type configfs (rw,relatime)
/dev/mmcblk0p1 on /boot/firmware type vfat (rw,relatime,fmask=0022,dmask=0022,codepage=437,iocharset=ascii,shortname=mixed,errors=remount-ro)
binfmt_misc on /proc/sys/fs/binfmt_misc type binfmt_misc (rw,relatime)
voltagex@kiwi:~$ cat /boot/config.txt
cat: /boot/config.txt: No such file or directory
```

`rpi-update` has worked fine, /dev/sda1 is mounted in /boot/firmware. I'm running the Jessie image from http://sjoerd.luon.net/posts/2015/02/debian-jessie-on-rpi2/ and I'm not sure what exactly made a mess of /boot

```
voltagex@kiwi:/boot/firmware$ cat /boot/firmware/config.txt
# For more options and information see
# http://www.raspberrypi.org/documentation/configuration/config-txt.md
# Some settings may impact device functionality. See link above for details

# uncomment if you get no picture on HDMI for a default "safe" mode
#hdmi_safe=1

# uncomment this if your display has a black border of unused pixels visible
# and your display can output without overscan
#disable_overscan=1

# uncomment the following to adjust overscan. Use positive numbers if console
# goes off screen, and negative if there is too much border
#overscan_left=16
#overscan_right=16
#overscan_top=16
#overscan_bottom=16

# uncomment to force a console size. By default it will be display's size minus
# overscan.
#framebuffer_width=1280
#framebuffer_height=720

# uncomment if hdmi display is not detected and composite is being output
#hdmi_force_hotplug=1

# uncomment to force a specific HDMI mode (this will force VGA)
#hdmi_group=1
#hdmi_mode=1

# uncomment to force a HDMI mode rather than DVI. This can make audio work in
# DMT (computer monitor) modes
#hdmi_drive=2

# uncomment to increase signal to HDMI, if you have interference, blanking, or
# no display
#config_hdmi_boost=4

# uncomment for composite PAL
#sdtv_mode=2

#uncomment to overclock the arm. 700 MHz is the default.
#arm_freq=800

```

XECDesign, 2015-12-24:
instead of `/dev/mmcblk0p1 on /boot/firmware` make it  `/dev/mmcblk0p1 on /boot` otherwise certain things won't work right, including rpi-update (I believe you're not running the firmware you think you're running). The bootloader only cared about what's on the fat32 partition. Running rpi-update in your case would've downloaded the files and put them somewhere where they don't do anything.

voltagex, 2015-12-24:
yep, I'll fix that mountpoint. I ended up wiping /boot and starting again. It's working now - I think it's that Debian image that had the non-standard structure.

closed 2015-12-24 by voltagex