RPi-Distro/raspi-config #6: non-interactive mode

nschloe, 2016-01-10:
I'm setting many (think 100+) raspberries, and I'd like to execute a bunch of raspi-config actions on all of them programmatically. Is it possible to rum raspi-config non-interactively? I'm particularly interested in expanding the file system and overclocking.

XECDesign, 2016-01-10:
Certainly, take a look here:
https://github.com/raspberrypi/rc_gui/blob/master/src/rc_gui.c#L21
`sudo raspi-config nonint do_expand_rootfs` and `sudo raspi-config nonint do_overclock %s`

You can find the usable values for %s here:
https://github.com/RPi-Distro/raspi-config/blob/master/raspi-config#L348

nschloe, 2016-01-10:
This command
```
sudo raspi-config nonint do_expand_rootfs
```
does start the interactive mode. What am I doing wrong?

XECDesign, 2016-01-10:
@spl237 Any idea?

spl237, 2016-01-10:
Off the top of my head, no idea - that should work in a non-interactive mode as that is the way rc_gui uses it, and I know that works, cos I've tried it. I'll have a look when I'm in the office tomorrow and see if I can repeat it locally.

spl237, 2016-01-10:
Ah - now I look at the code, it will use whiptail to show error messages if it can't expand the filesystem for some reason - these won't happen when run from rc_gui, as the error conditions are checked before raspi-config is called.

nonint was never really intended to allow functions to be called directly from the command line - it was a way for rc_gui to call them, which has the useful side-effect of making them (mostly) work from the CLI. I can fix this for a future release and get rid of the whiptail calls when do_expand_fs is called from the CLI, but there isn't a way to do it with the current version.

nschloe, 2016-01-10:
Good to know, thanks. The same is true for the second example
```
sudo raspi-config nonint do_overclock Pi2
```
by the way.

devgiants, 2016-01-11:
I just submit a pull request starting in that direction. Feel free to feedback!
Thanks,

Nicolas

spl237, 2016-01-12:
I've just tested these on a vanilla Jessie image from 21 November, the current release.

Both sudo raspi-config nonint do_expand_rootfs and sudo raspi-config nonint do_overclock Pi2 work non-interactively from the command line for me.

Are you using a recent Jessie image?

nschloe, 2016-01-12:
Well, I'm using Wheezy (via RetroPie).

XECDesign, 2016-01-12:
Wheezy is no longer supported.

nschloe, 2016-01-12:
The RetroPie folks are in the process of updating their stuff. I'll retest it then.

XECDesign, 2016-01-12:
Alright, but given that this particular feature has been tested and works for rc_gui, I'll close this issue. Thank you.

closed 2016-01-12 by XECDesign