RPi-Distro/raspi-config #9: Unusable if terminal window too narrow

navyscourge, 2016-01-16:
If using the x-terminal it is easy to set the terminal to be too narrow. Function calc_wt_size() detects a terminal that is less than 60 cols and sets the width to 80. This still leaves the terminal too narrow, and menu text columns are truncated and "squash" together. While this is not strictly unusable, it can be difficult to read the options.

If the terminal is too narrow, it should do one of:
a) automatically make the terminal wider (and redraw?),
b) continue and pop up a message that the terminal should be made wider,
c) pop up / display a message that the terminal should be made wider, and quit

I don't know if (a) or (b) are possible; perhaps (b) should recheck the width

XECDesign, 2016-01-19:
I am not sure I understand the issue. If the terminal window is too small to be convenient, why does the user need to be prompted to resize it? How would a user end up with a small terminal and be unaware that it can be resized?

navyscourge, 2016-01-20:
I did find that when I do resize the window, it does not redraw the window cleanly. I also might not make it wide enough, if using a GUI desktop with a terminal window and a browser. As this is likely to be the first thing a brand new user does, it seems clear from some forum posts that some of them do not understand many of the basic elements of functionality in computing.

XECDesign, 2016-01-20:
Why I don't think this is an issue:
1) To get into this situation, you'd need to know how to resize windows in the first place
2) Nobody has ever said "I've resized a window and don't know how to fix it"
3) Many programs are unusable if you make the window smaller than the content.
4) A lot of it comes down to behaviour of whiptail
5) raspi-config is a bash script which outgrew its purpose. It's already getting unwieldy and adding features like this just bloats it even more.
6) rc_gui should be used from the desktop instead.

I'll leave this open for now, in case someone has comments or suggestions.

spl237, 2016-01-23:
I agree with XECDesign's comments above - as per the other issue, rc_gui is now the recommended way to change Raspberry Pi settings. Further development on raspi-config will be functional only, not to improve usability.

closed 2016-01-23 by spl237