RPi-Distro/raspi-config #10: Jessie: raspi-config unable to change keyboard layout

astroleo, 2016-01-18:
After installing the Jessie image from raspbian.org, I wanted to change to a German keyboard layout using `sudo raspi-config` > `5 Internationalization Options` > `13 Change Keyboard Layout`. I had a German Macintosh USB keyboard connected to the Raspberry Pi 2.0 Model B. But when choosing this option, nothing happens and I am returned to the main menu of `raspi-config`. After exiting the config, I see these error messages on the console:
```
update-rc.d: warning: start and stop actions are no longer supported; falling back to defaults
update-rc.d: warning: start and stop actions are no longer supported; falling back to defaults
Reloading keymap. This may take a short while
```
I guess this is related to the change to `systemd`in Debian Jessie.

XECDesign, 2016-01-19:
Having trouble reproducing this issue.

The 'nothing happens' is the expected behaviour and those warnings aren't errors.

Is this under X? If so, do the changes 'stick' when using rc_gui instead?

astroleo, 2016-01-19:
Thanks for looking at this and sorry if I've not been precise enough. I wanted to change from standard US layout to German layout but after running raspi-config and selecting my German Macintosh keyboard, the raspi still uses the US layout (e.g. "z" is on "y" etc.). I was using raspi-config from a console started under X. I will try rc_gui asap.

XECDesign, 2016-01-19:
Another test: press ctrl-alt-f1, and check if the keyboard works as expected in the console.

I suspect the problem is that the way the keyboard is handled in X, is not the same it is done in the console. rc_gui is the recommended replacement, but raspi-config still exists for console users.

jonathanlyng, 2016-02-11:
I have the same issue. (Only using console) raspi-config gives the same error as OP. 
loadkeys fr gives the result: cannot open file fr

markwal, 2016-02-25:
I'm using the 2016-02-09-raspbian-jessie-lite.img headless over ssh, nox, and I reproduce the OP's results.  Choosing Change Keyboard Layout switches to console view, the update-rc.d messages appear and then it flips back to the raspi-config UI back to the Internationalization Options.  I expected it to ask me which keyboard layout I wanted.

XECDesign, 2016-02-25:
ssh was key to being able to reproduce this issue. Some environment variable seems to be making debconf think it's running in non-interactive mode, I think. I have tested that it works as expected from console and lxterminal, so I don't think this is a pi-specific issue. What does 'echo $TERM' say?

astroleo, 2016-02-25:
I tried it again today on a new Raspbian Jessie installation (2016-02-09 like @markwal). It showed the same error message, but after reboot it had accepted the new keyboard layout. I don't recall whether I have rebooted when I first reported the problem.

markwal, 2016-02-25:
raspi-config still just pops back to the Internationalization Options (with the update-rc.d warnings) after reboot.

```
TERM=xterm-256color
SHELL=/bin/bash
SSH_CLIENT=192.168.1.17 53154 22
SSH_TTY=/dev/pts/0
USER=pi
CSCOPE_EDITOR=vim
MAIL=/var/mail/pi
PATH=/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/local/games:/usr/games
PWD=/home/pi
LANG=en_US.UTF-8
SHLVL=1
HOME=/home/pi
LESS=-RXMF
LOGNAME=pi
VISUAL=vim
SSH_CONNECTION=192.168.1.17 53154 192.168.1.66 22
_=/usr/bin/env
```

XECDesign, 2016-03-08:
Unfortunately, since the problem is not with raspi-config, but with debconf, there's not much we can do on this end. I would recommend confirming that this happens under plain debian and then reporting the bug there.

closed 2016-03-08 by XECDesign