RPi-Distro/raspi-config #11: Changes to Navigation?

navyscourge, 2016-01-20:
I have looked at several issues, here and at the 'old' site at https://github.com/asb/raspi-config. The following Issues can be fixed by the suggested changes.

asb:11 - Raspi-config is not approachable by the blind...
Changed to use the a,b,c, menu options allowing hot keys (a, b, c,); is this more suited to people with poor vision, in addition to the colour-highlighted line? I don't know how to test this

asb:28 - raspi-config: Internationalisation Options - Selection
Using an Internationalisation menu item returns to the main menu, changed to return to same menu (also on Advanced menu)

asb:39 - display "Internationalisation Options" before "Change User Password"
If following the menu sequence, the keyboard layout should be corrected before setting the passsword; moved to option b

RPi-Distro:9 - Unusable if terminal window too narrow
If the width is less than 60, a message is given to select continue or quit. If too narrow the current raspi-config does not increase the terminal size? I am not sure how to test for a blank WT_WIDTH (if [ -z "$WT_WIDTH"...)

New: - pressing Escape just quits raspi-config
The pressing of Escape is detected, leaving a sub-menu or giving a confirmation message on the main menu before quitting

New: - Texts and comments
Some texts have been changed, some short comments added

Note: a new script is attached to this post. All update functionality has been removed; instead each option shows a message. This is to assist in testing the navigation. Lines beginning "#*nnn" preceding a function show the line number (nnn) in the original script that the function can be found. Run it in a terminal window from the GUI desktop to change the width of the screen

[nav_rconfig.txt](https://github.com/RPi-Distro/raspi-config/files/98190/nav_rconfig.txt)

spl237, 2016-01-23:
Going forward, the recommended use of raspi-config is via the rc_gui desktop GUI rather than the command-line raspi-config itself. raspi-config will be kept functionally up to date for people who don't want to use the GUI, but we're unlikely to do much in the way of usability improvements to it from now on.

closed 2016-01-23 by spl237

alecthegeek, 2017-01-15:
There is a use case for maintaining full functionality of the script with a full set of command line options. This will makes sense for people who need to automate the setup of Raspbian for educational or development purposes.

spl237, 2017-01-15:
The script will be maintained in future, and it does have a full set of command-line options - far more, in fact, than it had before the GUI was introduced, as the GUI uses said command-line options for all functionality.

dachshund-digital, 2019-02-17:
I can't believe that the use case for automation (via cli) is not seriously going to be supported.  I know of several environments that have many Pi devices, and drive the configuration via ansible, but ansible can't do everything with ease, where an answer file, or robust cli is the true need.  Please, seriously reconsider continuation of the cli aspect.  Sorry, but the GUI only model, is just plain wrong for serious use of Pi devices.  Are you not trying to encourage Pi devices at part of IoT?  There is just one valid use case for robust cli.

spl237, 2019-02-17:
Please read what I wrote above.

We are not going to stop supporting configuration via the CLI; in fact, the GUI application relies on the CLI configuration tool as a backend. The CLI configuration tool remains under continual development, as the number of ongoing check-ins to this project demonstrates.

What we are not doing is making *usability* changes to the CLI tool, as the GUI tool is where usability improvements are made. Functionality changes will continue to be made to the CLI tool as needed.