RPi-Distro/raspi-config #12: RFE - Expand filesize - option to reserve xx bytes/% of free space

juggledad, 2016-01-25:
I backed up an 8GB running Jessie install according to the directions on the foundation website. 
I then restored the image to a new 8GB sd card - according to the foundation directions. The new card fails to boot because there is a problem:
bad geometry: block count exceeds size of device
After MUCH time I discovered my 8GB SD cards were not all the same - of eight 8GB cards, I have:
2 - 7,906,263,040 GB
4 - 7,948,206,080 GB
2 - 8,026,849,280 GB

This can lead to much woe. (as can be seen by doing a google search)

Request For Enhancement
------------------------------------
Add an option to the raspi-config utility to expand to the full size of the sd card less x% or xx bytes.

If I had reduced by 2% I would have been able to restore any of those cards to any other of the cards with out the error.

Thanks

XECDesign, 2016-01-25:
raspi-config was a tool for beginners to get started quickly. I don't think this feature is something that such a tool needs. Instead. I would recommend a tarball backup instead or learn to resize partitions manually. Also, same reasons as in #11.

closed 2016-01-25 by XECDesign

juggledad, 2016-01-26:
If raspi-config is a tool for beginners, then the situation I explained needs to be addressed even more so. How's a beginner to deal with backups that FAIL? (ok the backup won't fail, but the restored SD card will fail). 

Knowing that all 8, 16 or whatever size SD cards don't have the same number of bytes would seem (to me) to make a fudge space on the card imperative especially for beginners relying on a tool such as raspy-config.

I've been dealing with computers for years and years and this took me several days to finally discover that some of my SD cards were smaller than others and that not all cards will restore properly. At first I thought my backups were corrupt.

Try explain all this to a beginner. You shouldn't have too, this tool for beginners should prevent this situation from happening to them.

XECDesign, 2016-01-26:
Yes, but I don't think "fudge space" is the answer. It's throwing away perfectly usable space and who knows what the lower limit for each size of sd card is for each manufacturer anyway? Then you'd also be adding a mystery option that's difficult to explain. On the desktop, you don't generally back up your data by writing out the whole raw disk, so I don't think it's good practice on the pi either. Make a tarball backup of what you need.

It's a problem with no easy answer. Each solution has its pros and cons.

@spl237 can you confirm that this isn't happening, or do you think it might be a good idea?

spl237, 2016-01-26:
It's counter-intuitive to throw away space on an SD card just in case it causes problems when copying the contents to another card.

A more sensible solution would be to use a backup/restore tool that checks and warns about targets being too small, or which does something intelligent like not copying unused blank space to the target in order to reduce the size required. Fudging every card we create isn't the answer.

XECDesign, 2016-01-26:
I haven't found a way to make such a tool work from windows. A livecd/usb like clonezilla might be the answer, but I haven't tried it for pi backups. Should work though.

juggledad, 2016-01-26:
Take a look at the documentation https://www.raspberrypi.org/documentation/linux/filesystem/backup.md about backing up the SD card. You are telling people to do it. 

I backup the whole SD card because I make changes to wpa_supplicant.conf for my hidden WiFi network, and add things into rc.local and install other packages so I backup the entire SD card.

XECDesign, 2016-01-26:
I understand that there's a problem with documentation. As a side note, it looks like this might be useful for windows users http://www.paragon-software.com/home/pm-personal/eshop.html. Gparted should work for linux users.

spl237, 2016-01-26:
To be fair, the documentation is describing how to back up and reinstall onto the same (or an identical) card, which will work without issue.

juggledad, 2016-01-26:
umm it says:
It may be sensible for you to keep a copy of the entire SD card image, so you can restore the whole SD card if you lose it or it becomes corrupt.

I bought eight Sandisk 8GB SD cards what would lead me to thinkthey weren't identical?

juggledad, 2016-01-30:
seems like people have been asking for this for a while:
https://www.raspberrypi.org/forums/view ... 76#p301976

If you search google using 'rpi backing up sd card' almost everyone says the same thing 'use dd to back and restore the card' but this causes problems if the new card is smaller than the old. Lots and lots of people have run into this issue over the years, so if rapsi-config can't/won't be changed, then the foundation owes it to the public to show (with step by step instructions) how to properly backup an sd card and restore it to a new sd card (which could be smaller) using OSX, Linux or Windows - remembering that not everyone is a Linux Guru.

spl237, 2016-01-30:
Restoring a dd'ed backup to a smaller card is possible, but is a non-trivial operation. If you want / need to do it, just Google "use dd to restore to smaller disk" or similar and read the advice on the numerous relevant pages which come up. This is not a Raspberry Pi-specific question; it is a generic Linux issue. There are also several alternative backup options for Linux which you can investigate if dd doesn't do what you need. 

Our advice would be to ensure you only ever restore to a card the same size or larger than the one you backed up from, which is what most people would end up doing, given that the purpose of a backup is to recover from something that corrupts the disk you are using - by definition, you can then restore to the disk from which you backed up, which will be the right size.

XECDesign, 2016-01-30:
Since this isn't a raspi-config issue, please keep further discussion to [your forum thread](https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=134070) or open a documentation issue.

If you have a contribution to make to the documentation, a pull request would be welcome. Thank you.

juggledad, 2016-01-30:
> Since this isn't a raspi-config issue, please keep further discussion to your forum thread or open a documentation issue.

I respectfully disagree. Raspi-config is the tool that is used to configure a SD card for usage and a forum thread does not get the attention of the people in charge of raspi-config. 

> raspi-config was a tool for beginners to get started quickly
Isn't this what the Pi developed for? Beginners? To get beginners using and hacking things? Why else would there now be a NOOBS install?

Since the raspi-config is beginners why not build in a fudge? This would eliminate the angst that beginners (and others) get trying to clone their work. By your logic, power users sould know how to reclaim the extra space and NOOBS aren't going to miss it.

> On the desktop, you don't generally back up your data by writing out the whole raw disk
I do! I regullary make a clone of my entire disk incase the disk dies, just what people do when they try to clone their SD cards.

> I don't think it's good practice on the pi either. Make a tarball backup of what you need.
Fine, please provide me with (or point me to) step by step instructions to make a tarball backup and step by step instructions to recover it to a another random SD card of the same size (8G to 8G or 16G to 16G) - using OS X - noting that all 8G cards are not the same size.

If you can show me how, and the cloned SD card works, I'll be happy to drop this issue and write it up so beginners will know how to clone their SD cards.