RPi-Distro/raspi-config #14: Support HifiBerry Digi+

PanderMusubi, 2016-02-10:
Please support HifiBerry Digi+ S/PDIF audio via Toslink by adding or removing the following line in /boot/config.txt

    dtoverlay=hifiberry-digi

See also https://www.hifiberry.com/digiplus/

Additionally, alter values in /etc/asound.conf to set this as default audio card

    pcm.!default  {
     type hw card 0
    }
    ctl.!default {
     type hw card 0
    }

The values are probably not the default as shown above but can be querried with

    aplay -l

and

    aplay -L

If needed, I can provide output of those command and test raspi-config

lurch, 2016-04-01:
There are many add-on audio cards for the Raspberry Pi, and as an outside observer I wouldn't expect raspi-config to add support for one specific make and model of audio card...?

XECDesign, 2016-04-01:
Thanks lurch. We can't support every third party card in raspi-config.

closed 2016-04-01 by XECDesign