RPi-Distro/raspi-config #15: Disable / enable WiFi power saving

PanderMusubi, 2016-02-10:
Please offer support for disabling and enabling WiFi power saving. This involves the creation of a file called /etc/modprobe.d/8192cu.conf

To disable, add the following line.

    options 8192cu rtw_power_mgnt=0 rtw_enusbss=0 rtw_ips_mode=1

To enable, remove the line above.

Implementing this prevents systems locking down on remote access because power saving is disabling WiFi, rendering the system unapproachable.

peterblazejewicz, 2016-02-13:
The WiFi (of any kind) is not part of standard hardware configuration? Why it should be configured via this tool?

PanderMusubi, 2016-03-18:
Should it be since Raspberry Pi 3?

There is more in raspi-config that is not standard RPi hardware. Implementing this feature will allow users, mainly less advanced users, to easily stay connected with their RPi instead of loosing connection of a reason which is hard for them to duplicate when searching for the cause using the device with keyboard etc connected (as interaction keeps it alive).

peterblazejewicz, 2016-03-18:
> Should it be since Raspberry Pi 3?

My comment was from 13th, obviously I was not aware about WiFi.

PanderMusubi, 2016-03-18:
;-)

lurch, 2016-04-01:
> Should it be since Raspberry Pi 3?

The WiFi chip on a Raspberry Pi 3 is a BCM43438 and therefore the 8192cu-specific options mentioned in the first post would have no effect.

It's not something I've experimented with myself, but https://www.raspberrypi.org/forums/viewtopic.php?t=138731&p=921437 and http://raspberrypi.stackexchange.com/questions/43473/does-the-bcm43438-wifi-chip-in-raspberry-pi-3-support-power-management-setting suggest some ways of modifying the WiFi power-saving settings on a Pi3.

XECDesign, 2016-04-26:
The driver for the official dongle already disables power saving until issues with it are resolved. Since this isn't really an option we'd expect people to toggle on and off often (if at all), it's not something that should go into raspi-config.

closed 2016-04-26 by XECDesign

PanderMusubi, 2016-04-26:
Thanks for the elaboration.