RPi-Distro/raspi-config #16: Limit installation of related packages

PanderMusubi, 2016-02-10:
Please support limiting installation of related packages that are pulled in via recommends and suggests dependencies. This concerns the file /etc/apt/apt.conf and the following lines should be added for limiting the amount of packages that are pulled in.

    APT::Install-Recommends "false";
    APT::Install-Suggests "false";

Removing these lines will enable installation or related packages.

Implementing this will allow for a less disk usage on these systems that are aimed at minimal hardware requirements. The packages deciding on that is recommends and suggests do this with a server, desktop or laptop in mind, and not more limited device as an RPi.

lurch, 2016-04-01:
IMHO this issue should be closed here, and re-opened at https://github.com/RPi-Distro/repo/issues

XECDesign, 2016-04-01:
The default settings are appropriate. Not installing recommends on all packages would definitely stop things from working the way people expect them to. We also make use of recommends in our own packages. apt-get has the --no-install-recommends option for the exceptions.

closed 2016-04-01 by XECDesign