RPi-Distro/raspi-config #18: Error enabling desktop GL driver

PanderMusubi, 2016-02-10:
When choosing

    AA GL Driver    Enable/Disable experimental desktop GL driver 

and xcompmgr is not installed, an error is shown and the instruction to install that package.

Raspi-config could easily offer the user to install that package for him or her.

Fixing this will make the configuration process more easy for the end-user.

PS  Pershaps rename this option so the user know he or she is using a closed or open driver, because that is what really happens, regardless that it is experimental (as much is).

lurch, 2016-04-01:
> Pershaps rename this option so the user know he or she is using a closed or open driver

It's not as simple as open vs. closed source - each driver offers different features, see #21

XECDesign, 2016-04-01:
Enabling that option can result in a system that doesn't boot properly and difficult for beginners to recover from. If the user has read the blog post about it and followed the instructions there, they would've installed xcompmgr. We don't want to make it easy for people to accidentally enable this feature without knowing what they're installing and why.

closed 2016-04-01 by XECDesign

PanderMusubi, 2016-04-01:
Perhaps better to keep that option in the development version of raspi-config for now then. Otherwise you offer something you know it won't work.

lurch, 2016-04-01:
> Otherwise you offer something you know it won't work.

It *might* not work - that's something very different from "you know it won't work".
AFAIK the point is to make it easy for advanced / adventurous users to enable the OpenGL feature, so they can help test it and provide feedback to RPF, so that they can make it more stable in future.
https://www.raspberrypi.org/blog/another-new-raspbian-release/