RPi-Distro/raspi-config #23: Need option to change UART allocation and core_freq

rurwin, 2016-04-07:
See: https://www.raspberrypi.org/forums/viewtopic.php?f=44&t=142112

The GPIO UART on the Pi3 is badly broken, as in unusable. There seem to be three options, none of which is obviously best.

- As is -- bluetooth works, but the GPIO UART doesn't.
-     Both bluetooth and GPIO UARTs work but the GPU core frequency is throttled back to its minimum.
-     The GPIO UART works but there is no Bluetooth UART.

Setting up these options is complex and fiddly, especially the third which involves DT overlays. On the other hand it seems to be perfect for a new raspi-config option.

popcornmix, 2016-04-07:
Latest firmware/kernel (probably only rpi-update currently) disables uart on Pi3 by default, and supports option ``enable_uart=1``.
With ``enabled_uart=1`` you will get a usable uart (most likely by core_freq overclock being disabled).

XECDesign, 2016-04-26:
Should be fixed by 8af52bed399b99e7dbd682c090649bdfe28af4ca

closed 2016-04-26 by XECDesign