RPi-Distro/raspi-config #24: gpu_mem adjustments are overridden by gpu_mem_1024/512/256

dankcushions, 2016-04-20:
Hi! The current gpu_mem adjustment option only seems to edit the gpu_mem = X entry on config.txt. Fair enough, but this value is overridden by the gpu_mem_256/512/1024 (depending on what pi you have) entries, as per http://elinux.org/RPiconfig#Memory

I believe these entries are not in config.txt by default (see https://www.raspberrypi.org/documentation/configuration/config-txt.md), but I think perhaps there should be a warning, if they are present?

XECDesign, 2016-04-21:
There are a number of ways to override options which raspi-config sets, and I don't think we can check for all of them.

On one hand, if you're using gpu_mem_x, you probably shouldn't be using raspi-config to adjust the memory split. On the other hand, I think I've seen a similar issue reported before, so maybe this particular case is worth addressing.

@spl237 Thoughts?

spl237, 2016-04-21:
I've modified both raspi-config and rc_gui so that they read in the value currently in effect, including the gpu_mem_<size> values. If either raspi-config or rc_gui is used to set a value, it is written to the global gpu_mem parameter only, and any gpu_mem_<size> parameters are commented out; a warning dialog is shown in raspi-config if this happens.

closed 2016-04-21 by spl237

dankcushions, 2016-04-21:
Oh wow, great thanks! :+1: