RPi-Distro/raspi-config #25: no docs for non-interactive use

CarlFK, 2016-04-22:
I want to use the non-interactive things, but I am having to trace though the bash script to figure out how it works.

Can I get some docs on how this works?
examples examples would be great.

my use case:
I want to document how to setup a pi for this class:
https://github.com/mechanicalgirl/young-coders-tutorial

like start with current raspbian, and run these raspi-config commands to set the following:

raspi-config
expand fs
boot into gui
set locale to en_US
keyboard-layout
timezone

XECDesign, 2016-04-22:
Updating documentation is on the todo list, but for now, take a look here:
https://github.com/RPi-Distro/rc_gui/blob/master/src/rc_gui.c#L21

Romeo-Golf, 2016-12-18:
I suspect many others, myself included, would find this useful :)

hxmuller, 2017-01-28:
I believe what  **CarlFK** _really_ wants to do, is to run raspi-config _non-interactively_.

You could add this to the documentation, but it's really not just a raspi-config thing, but a thing you may want to do for _any_ program that requests user input, that does not have an option to run non-interactively, that you want to automate.

The solution is to write an **expect script** to run the application non-interactively.  To do this **without** learning how to write an _expect script_, on any Debian based installation, run the following:

$ sudo apt-get install expect

Additionally, if you are not installing recommends, then you also need to install tcl8.6 for autoexpect to work:

$ sudo apt-get install tcl8.6

Then run the following to automagically create the desired expect script:

$ sudo autoexpect raspi-config

Answer the questions to get the results you want. If you make an error during the configuration, merely delete the automatically generated script.exp file and rerun the above command.

You can rename script.exp file to something more descriptive. Then you only have to ensure the base image that the students are going to be using, include expect, because it is not included in either the official desktop or minimal image.

Now to perform the configuration automatically without user input, you merely run the automatically generated script, script.exp or whatever you may have renamed it to:

$ sudo ./script.exp

expect is not hard to learn, but you do not need to learn it unless you want to do something more complex than autoexpect can handle. But autoexpect should be just fine for the use case CarlFK is interested in.

CarlFK, 2017-01-28:
> that does not have an option to run non-interactively, 

It does seem to have, or want to have, the option:

INTERACTIVE=True
https://github.com/RPi-Distro/raspi-config/blob/master/raspi-config#L6

hxmuller, 2017-01-28:
CarlFK

True, but a review of the current state of the raspi-config shell code shows that the only option that you desire that can be run non-interactively is --expand-rootfs, all of the other configuration changes you desire are implemented as 'user input required' functions.

So to get just that single configuration option done, you would run:

$ sudo raspi-config --expand-rootfs

To automate **all** of your desired configuration changes, then you will have to consider the previous suggestion or something similar.

XECDesign, 2017-03-03:
This issue should probably be opened on the documentation repo.

closed 2017-03-03 by XECDesign