RPi-Distro/raspi-config #28: "I3 Change Keyboard Layout" won't work after "apt-get dist-upgrade"

Saftwerk, 2016-06-11:
I did this in sequence:
**raspi-config** (<-- here I can use 'Change Keyboard Layout' without problems)
**apt-get update**
**apt-get dist-upgrade**
**raspi-config** (<-- here I get bad messages when I select 'Change Keyboard Layout')

This is the bad message:
```
update-rc.d: warning: start and stop actions are no longer supported; falling back to defaults
update-rc.d: warning: start and stop actions are no longer supported; falling back to defaults
Reloading keymap. This may take a short while
```

Conclusion:
**raspi-config** needs to be fixed to work with latest updates.

OS: Raspbian Jessie
used hardware: Raspberry Pi 3 Model B 1GB RAM

Saftwerk, 2016-06-11:
After a closeer look I found this command invokes the bad massege:

dpkg-reconfigure keyboard-configuration
see file: https://github.com/RPi-Distro/raspi-config/blob/master/raspi-config#L298

the similar commands are working fine:
dpkg-reconfigure tzdata
dpkg-reconfigure locales

XECDesign, 2016-06-12:
Does something not work? I don't see any errors or bad messages there.

Saftwerk, 2016-06-13:
Well, there is no Error message.
But the dialogs to change the keyboard will appear no more.

This is missing:
[https://s.blogcdn.com/www.engadget.com/media/2012/08/setkeyboardopt3.png]()

XECDesign, 2016-06-13:
Are you running raspi-config locally or through ssh?

Saftwerk, 2016-06-13:
I don't remember accurate, because I use my HDMI monitor with dual inputs and I switched often between them.

All I can say is: that packages below were updated and then the dialogs no longer appears.
If it is necessary, then I could do it all over again, but one at a time.
(but i can do this only at the weekend)

**apt-get dist-upgrade**
The following packages will be upgraded:
  apt apt-utils bind9-host cpio dmsetup dpkg dpkg-dev firmware-atheros firmware-brcm80211 firmware-libertas
  firmware-ralink firmware-realtek git git-core git-man gnupg gpgv initramfs-tools libapt-inst1.5 libapt-pkg4.12
  libarchive13 libbind9-90 libc-bin libc-dev-bin libc6 libc6-dbg libc6-dev libdevmapper1.02.1 libdns-export100
  libdns100 libdpkg-perl libexpat1 libgd3 libhogweed2 libidn11 libirs-export91 libisc-export95 libisc95
  libisccc90 libisccfg-export90 libisccfg90 liblwres90 libnettle4 libpcre3 libraspberrypi-bin libraspberrypi-dev
  libraspberrypi-doc libraspberrypi0 libssh2-1 libssl-dev libssl-doc libssl1.0.0 libsvn1 libsystemd0 libtalloc2
  libtasn1-6 libtasn1-6-dev libtasn1-doc libudev1 libvpx1 libwbclient0 libxapian22 libxml2 locales
  multiarch-support nettle-dev openssh-client openssh-server openssh-sftp-server openssl perl perl-base
  perl-modules python-rpi.gpio raspberrypi-bootloader raspberrypi-sys-mods samba-common ssh subversion systemd
  systemd-sysv tzdata udev
83 upgraded, 2 newly installed, 0 to remove and 0 not upgraded.
Need to get 122 MB of archives.

XECDesign, 2016-06-15:
If you were running it through ssh, that's the expected behaviour. As you saw, all raspi-config does here is call dpkg-reconfigure, which comes from Debian and is not modified by us.

Saftwerk, 2016-06-16:
OMG! Yes, You're right i tried it with a fresh image.
using HDMI: Change Keyboard Layout works as expected.
using ssh(putty): Change Keyboard Layout wil not show any dialog.

Case closed
and thanks You for support.

closed 2016-06-16 by Saftwerk

EveryOtherUsernameWasAlreadyTaken, 2016-08-29:
@XECDesign "that's the expected behaviour" - expected by whom? I spent hours trying to fix it. And why wouldn't one want to configure a keyboard via SSH? The minimum appropriate would be an error message that it is not possible via SSH and that this is the defined behavior that will not be changed.

XECDesign, 2016-08-29:
@EveryOtherUsernameWasAlreadyTaken This is behaviour inherited from upstream Debian.
https://www.debian.org/Bugs/Reporting

The warning message is a good idea, thanks.