RPi-Distro/raspi-config #29: Disabling Serial removes /dev/ttyACM0

gohai, 2016-06-18:
Disabling the serial terminal via raspi-config also removes the device node `/dev/ttyACM0`, which is different to how this worked previously, where it remained and could be used by user-space applications.

This is on Raspbian 2016-05-27.

gohai, 2016-06-18:
PS: If it makes a difference: I was testing on a Raspberry Pi Zero.

gohai, 2016-06-18:
Think it's because of this line: https://github.com/RPi-Distro/raspi-config/blob/master/raspi-config#L709

dgordon42, 2016-07-08:
Answering "NO" to the question "Would you like a login shell to be accessible over serial?" should just disable the getty, not the port itself.
It's causing some confusion, see this example at [https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=153514](url)
The port just quietly disappears, and a reboot does not bring it back, even with "enable_uart=1" in '/boot/config.txt'
Thanks,
Dave.

Romeo-Golf, 2016-12-18:
Would be nice if this could be fixed as I would imagine the primary use case for disabling the console access via the serial port is to use it for some other user space application :)

itemir, 2017-01-10:
I agree this is confusing. There seems to be multiple issues at play here. But first thing to confirm is the device in question. It should be  /dev/ttyAMA0, not /dev/ttyACM0 as mentioned in the title. Is that the case? Assuming it is, a couple of additional notes:

 First of all, which is only tangentially related, get_serial does not seem to be complete/correct:

```
get_serial() {
  if grep -q "ttyAMA" /proc/cmdline ; then
    if grep -q -E "^enable_uart=0" $CONFIG; then
      echo 1
    else
      echo 0
    fi
  else
    if grep -q -E "^enable_uart=1" $CONFIG; then
      echo 0
    else
      echo 1
    fi
  fi
}
```

This code practically checks the presence of enable_uart in /boot/config.txt and if it is set to 1 it assumes console is enabled, if it is 0, console is disabled (it also does it in a confusing way and not sure why. If someone knows the history of this piece of the code, I would be interested in learning the logic that went in. grepping for ttyAMA in /proc/cmdline does not seem have a practical impact). 

Independent of it though, this code only checks if UART is enabled, does not check if it is attached to console properly (it does some kind of a check but seems to be indifferent about the result, which is fairly confusing). 

Bottom line is, UART can be enabled but console can be disabled, which is a typical configuration if you want to use UART for your own purposes anyways. In that scenario, this function returns an incorrect status.

Independent of above, there is the code that actually disables it:

```
    sed -i $CMDLINE -e "s/console=ttyAMA0,[0-9]\+ //"
    sed -i $CMDLINE -e "s/console=serial0,[0-9]\+ //"
    set_config_var enable_uart 0 $CONFIG
    STATUS=disabled
```

The first two lines with sed should be enough, as they remove the console= setting from /boot/cmdline.txt, while still retaining console=tty1. However it goes further and sets enable_uart to 0, and that should not required for disabling the console. That is in fact causing /dev/ttyAMA0 to disappear. I have tested this on a Pi3 and it is acting as such. I think it will will behave similarly on other platforms but testing would be helpful.

Romeo-Golf, 2017-01-10:
I can confirmt he relevant serial interface is /dev/ttyAMA0, at least on Raspberry Pi 3...

And I agree that since the question asked to the user is "Would you like a login shell to be accessible over serial?" then if they answer "no" then it should only remove the console= lines and leave the current value of enable_uart= alone...

This may not be what everyone wants/needs, if so then perhaps if the user answers No to ""Would you like a login shell to be accessible over serial?" then they could be asked a follow on question of "Would you also like to disable the serial port(s)?"

Regards.

spl237, 2017-01-10:
Code now modified to first ask if the serial shell is wanted; if it is not, asks if the hardware is still wanted. 

Noninteractive use from command line and call from Raspberry Pi Configuration GUI app still switches both simultaneously to reduce number of user options in desktop.

closed 2017-01-10 by spl237

gohai, 2017-01-10:
@spl237 Thanks, but curious: what's the requirement for two options? Why would you *not* want to have the device, when disabling the serial shell?

Asking because I'd love to be able to give people a quick way to use the Serial library with Processing, and so far all other (relevant) configuration options could be accomplished using the Raspberry Pi Configuration GUI app.

spl237, 2017-01-10:
Some people may want to use the UART for a different purpose, and hence we need to be able to unassign it from the serial port.

gohai, 2017-01-10:
@spl237 But users that don't want the login shell _and_ don't want the `ttyACM0` device node would be the special case, I presume? Couldn't the logic be reverted, that for those special use cases, one would need to use `raspbi-config`, but that the GUI app enables `ttyACM0` when disabling the login shell?

XECDesign, 2017-01-10:
@gohai I suspect that users might also want to disable it entirely to be able to use the pins as GPIOs. I don't know if it's necessarily true that that case is less likely than using serial for anything other than a login shell.

gohai, 2017-01-10:
@XECDesign Yet in all pinouts I have seen the pins are distinctly marked as UART. I would only attempt to use those pins when I have no "vanilla" pin left, and then I also wouldn't expect those pins to be handed to me easily. But using the serial port to be talking to some GPS module, receipt printer, etc...

XECDesign, 2017-01-10:
True

spl237, 2017-01-10:
Whichever way we do this, some people will want it the other way. If you want to configure it differently from the command line, it's simple enough to re-enable the serial port just by a direct call to the raspi-config function that sets config variables:

sudo raspi-config nonint do_serial 1
sudo raspi-config nonint set_config_var enable_uart 1 /boot/config.txt

gohai, 2017-01-10:
Thanks @spl237. I appreciate your message and effort to keep things as clear as possible. I am working on a book concerning the Pi currently, that's why I am perhaps longing for a very stable & self-describing configuration mechanism.

gohai, 2018-03-16:
@spl237 Does enabling the UART still comes with a performance hit on recent versions of Raspbian? Is this the same on the 3+? Thank you!