RPi-Distro/raspi-config #30: Support setting config_hdmi_boost

PanderMusubi, 2016-09-20:
Plesae, support setting config_hdmi_boost (with the proper warning or info such as only use this when HDMI output is not strong enough and use the lowest  level possible).

    CONFIG_HDMI_BOOST
    
    Configures the signal strength of the HDMI interface; the default value is 0 and the maximum is 7. Try 4 if you have interference issues with HDMI.

from https://www.raspberrypi.org/documentation/configuration/config-txt.md

Romeo-Golf, 2017-01-10:
I guess this could be useful in some use cases....

PanderMusubi, 2017-01-10:
Especially when using a HDMI splitter cable.

XECDesign, 2017-01-10:
This is an option that should not be touched by the average user and can have negative consequences. It's outside of the scope of what raspi-config is for.

closed 2017-01-10 by XECDesign