RPi-Distro/raspi-config #32: Add no country restriction for wifi

PanderMusubi, 2016-10-01:
Please add no country restriction to choose from when setting country restriction for wifi. This might be relevant in a shielded laboratory setup to test certain wifi devices or in international waters, antartica or on a space station. There are perhaps more reasons to set, or go back to, no country restriction.

The option could be called default or none. Perhaps call it `none` and use that as default.

StatelessCat, 2016-11-13:
I need this too, for educational/laboratory purposes.

XECDesign, 2017-01-10:
This is a little outside of the scope of raspi-config.

closed 2017-01-10 by XECDesign

PanderMusubi, 2017-01-10:
If you do not set it, it is undefined. Once you set it via raspi-config, you cannot unset it in raspi-config. Offering unsetting it it will make raspi-config fully symmetric in dealing with this setting.

PanderMusubi, 2017-01-10:
If you do not set it, it is undefined. Once you set it via raspi-config, you cannot unset it in raspi-config. Offering unsetting it it will make raspi-config fully symmetric in dealing with this setting. This symmetry is found when manipulating other settings via raspi-config.

Romeo-Golf, 2017-01-10:
I agree, just as its good manners to close the door on your way out this script should unset any variables it has set :)

XECDesign, 2017-01-10:
Interesting point.

We ship with it set to GB. https://github.com/RPi-Distro/pi-gen/blob/dev/stage2/02-net-tweaks/files/wpa_supplicant.conf