RPi-Distro/raspi-config #34: VNC Enable Does Not Work

elirab52, 2016-10-08:
I select VNC Enable on Pi Configuration screen, select ok, but it defaults back to disable.
Please advise

carriba, 2016-10-28:
Manually install a VNC server as exemplary described with the [Installing TightVNCServer](https://www.raspberrypi.org/forums/viewtopic.php?p=1049099#p1049099) article and then restart the LightDM process.

Maybe the "do_vnc" code section of the "raspi-cionfig" tool can be amended accordingly in a future release to

1. install a VNC server application if not already installed

2. simply update the "/etc/ligthdm/lightdm.conf" with the VNC server settings. 

LightDM will take care of starting and shutting down the VNC server process.

XECDesign, 2016-10-28:
I believe raspi-config is meant to be used with realvnc, not tightvnc.

XECDesign, 2016-10-28:
@spl237 any idea?

carriba, 2016-10-28:
OK, I tried it with the preinstalled RealVNC server package from the September 2016 Raspbian Jessie, and it failed beacuase of apparently no RealVNC Server license, sic! 

I had the same issue connecting to the headless RPi device with a VNC client, and thus manually started the preinstalled VNC server daemon and am getting the following output:

pi@raspberrypi# sudo systemctl start vncserver-x11-serviced.service
Job for vncserver-x11-serviced.service failed. See 'systemctl status vncserver-x11-serviced.service' and 'journalctl -xn' for details.
pi@raspberrypi# sudo systemctl status vncserver-x11-serviced.service
● vncserver-x11-serviced.service - LSB: VNC Server in Service Mode daemon
   Loaded: loaded (/etc/init.d/vncserver-x11-serviced)
   Active: failed (Result: exit-code) since Fri 2016-10-28 17:23:24 CEST; 16s ago
  Process: 2950 ExecStart=/etc/init.d/vncserver-x11-serviced start (code=exited, status=127)

Oct 28 17:23:24 raspberrypi vncserver-x11-serviced[2950]: /etc/init.d/vncserver-x11-serviced: 29: /etc/init.d/vncserver-x11-serviced: vnclicense: not found
Oct 28 17:23:24 raspberrypi vncserver-x11-serviced[2950]: vncserver-x11-serviced: No VNC licenses are installed. To enable connections,
Oct 28 17:23:24 raspberrypi vncserver-x11-serviced[2950]: apply a license with vnclicense.  If you apply a Free license and have not
Oct 28 17:23:24 raspberrypi vncserver-x11-serviced[2950]: yet specified a VNC password, please also run as root
Oct 28 17:23:24 raspberrypi vncserver-x11-serviced[2950]: `vncpasswd /root/.vnc/config.d/vncserver-x11'.
Oct 28 17:23:24 raspberrypi vncserver-x11-serviced[2950]: /etc/init.d/vncserver-x11-serviced: 34: /etc/init.d/vncserver-x11-serviced: vncserver-x11-serviced: not found
Oct 28 17:23:24 raspberrypi systemd[1]: vncserver-x11-serviced.service: control process exited, code=exited status=127
Oct 28 17:23:24 raspberrypi systemd[1]: Failed to start LSB: VNC Server in Service Mode daemon.
Oct 28 17:23:24 raspberrypi systemd[1]: Unit vncserver-x11-serviced.service entered failed state.

Having manually installed the RealVNC package as outlined with the [Installing or upgrading VNC Server manually](https://www.realvnc.com/docs/raspberry-pi.html#raspberry-pi-legacy) article, it seems that I can start the VNC deamon as the built-in license from RealVNC has been applied. See the following output:

pi@raspberrypi# sudo /usr/bin/vnclicense -list
License key 1, RaspberryPi
- Serial number: ...
- Location: /etc/vnc/licensekeys/vncserver
- Issue number: 0
- Desktops: 10
- Support and upgrades: - Expiry undefined
- Features provided:
   - 128-bit AES encryption
   - Chat
   - File transfer
   - Optimized performance
   - Printing
   - System authentication
   - VNC authentication

Thus, the "raspi-config" script ought to provide a hook to check whether the installed RealVNC application package is licensed or not.

To check for a valid license key, simply fire the command "/usr/bin/vnclicense -check"" that shall return "0" if a valid license key is installed.

spl237, 2016-10-28:
Raspi-config is only intended to switch the RealVNC server on or off, not other VNC servers. It already checks to see if the RealVNC server package is installed and disables the switch if not. The RealVNC server package automatically installs a valid licenses - so the interesting question here is how that process failed. I don't think there's a strong argument for modifying the behaviour of raspi-config - if you have the package installed, it should be licensed. If we see this happening more often, I'll raise it with RealVNC.

carriba, 2016-11-03:
OK & thanks for the answer. 

In the meantime I got a VNC session working using the RealVNC Server, see the posting "[Connnecting to RealVNC Server](https://www.raspberrypi.org/forums/viewtopic.php?p=1062552#p1062552)" for details.

closed 2017-01-10 by XECDesign