RPi-Distro/raspi-config #36: Overclocking Option not working for Rasperry Pi Zero

BouncingBit, 2016-11-01:
I've tried to change the Overclock option when using my new Rasperry Pi Zero. I know, that the CPU by itself is already clocked at 1000 MhZ, thus I guess this option doesn't make sense for the Pi Zero anyway, right? This isn't a bug per se, but should this option be disabled for the Pi Zero?

XECDesign, 2016-11-01:
Correct, the options are only meant for pi 1 and pi 2.

BouncingBit, 2016-11-02:
Okay, shall I close this issue then, or is there a possiblity to disable the option for the Rasperry Pi Zero programatically?

XECDesign, 2016-11-02:
What do you mean by that? The option should already say that the board can't be overclocked. Are you seeing something different?

BouncingBit, 2016-11-02:
Sorry. You're right. There indeed is an appropriate message "This Pi cannot be overclocked" and then "There was an error running option 8 Overclock". Hm... I thought I saw some screen where the clocking speed was displayed as 700 Mhz... strange...

closed 2016-11-02 by BouncingBit