RPi-Distro/raspi-config #38: there is no easy way to disable (or configure at all) the screensaver in the GUI

FlorinAndrei, 2016-11-15:
What the title says. Changing any setting related to the screensaver - there's no way to do it in the GUI. Something as simple as disabling it - no way either.

This is a very, very common configuration task. It must be included in the GUI configurator. At the very least, allow a simple enable/disable toggle, with the default set to whatever you prefer.

Please note that this applies to the screensaver in both the graphical mode, and in the text (console mode). The settings are likely very different. I feel the graphical mode has a higher priority, but eventually both should be addressed.

richhaynes, 2017-03-08:
Are you referring to the screen going black as the screensaver? If so, from what i have learned is that it is more akin to a power saving feature than a sceensaver.

A normal computer has power saving tech that can tell a screen to turn off when its not been used for X minutes. Since the RPi is not a normal computer, it cannot put a screen to sleep like a normal computer and so it does the next best thing which is blanking the screen (i.e. output all black). This is because a black image uses less power than a bright desktop image. I believe this is done at the hardware level rather than software so it cant be configured that easy if at all.

If you are away from the RPi for ten or so minutes, it makes sense to do this. You only have to shake your mouse of press space on your keyboard to wake it and thats a small bit of inconvenience to save a chunk of power consumed by power-hungry displays :)

malaire, 2017-09-07:
It does NOT make sense to turn off display. I might be doing long build which I want to monitor without needing to move mouse every 10 minutes.

XECDesign, 2020-01-07:
Option to disable screen blanking added.

closed 2020-01-07 by XECDesign