RPi-Distro/raspi-config #41: Ability to call any of the raspi-config commands non interactivly

Romeo-Golf, 2016-12-24:
I gather there is some support for a limited subset of the commands available on the raspi-config GUI; however it would be nice if this could be made available for *all* commands and functionality directly from the CLI so that the large number of people doing unattended installs can leverage this functionality to ensure these changes are applied consistently to doing the same thing via the menus...

Also it would be great if these could be documented somewhere other than the code ;)

Romeo-Golf, 2016-12-24:
There seems to have been some previous threads that indicated there is demand for this, sadly despite some commits and suggestions to remove the whitetails these all seem to have ended up being closed:

https://github.com/RPi-Distro/raspi-config/issues/6

https://github.com/RPi-Distro/raspi-config/pull/7

spl237, 2017-01-10:
The only function that wasn't callable by the CLI was audio output selection - this has now been added for the next release.

closed 2017-01-10 by spl237