RPi-Distro/raspi-config #42: Support setting max_usb_current

PanderMusubi, 2017-01-06:
Please, support setting max_usb_current (with the proper warning or info such as only use this when USB is lacking enough power).

    MAX_USB_CURRENT
    
    Configures the available current over USB; the default value is 0 for 600 mA. A value of 1 sets  it to 1.2 A.

From https://raspberrypi.stackexchange.com/questions/27708/is-setting-max-usb-current-1-to-give-more-power-to-usb-devices-a-bad-idea

See also https://github.com/RPi-Distro/raspi-config/issues/30

Romeo-Golf, 2017-01-10:
Does the following still hold true:

"This feature is only available on the B+ and Pi model 2 (at the time of this writing)"

PanderMusubi, 2017-01-10:
I don't know. What would be a good place to find out or ask?

PanderMusubi, 2017-01-10:
I don't know. What would be a good place to find out or ask? PS This is useful when having e.g. only one USB device, such as a camera, attached that needs a bit more power in order to start working.

Romeo-Golf, 2017-01-10:
This thread was the top search result for "raspberry pi 3 max_usb_current=1"

https://www.reddit.com/r/raspberry_pi/comments/4b2enb/psa_raspberry_pi_3_provides_12a_usb_current_by/

XECDesign, 2017-01-10:
It has been removed as a firmware option. The default now is to always allow max USB current.

closed 2017-01-10 by XECDesign

Romeo-Golf, 2017-01-10:
Does the "default now is to always allow max USB current" apply to all previous generations of Raspberry Pi, or does this script only support the Raspberry Pi 3?

XECDesign, 2017-01-10:
This applies to all models where max_usb_current had any effect (B+, A+ and later).

Romeo-Golf, 2017-01-10:
XECDesign, can you advise where it is documented that Raspberry Pi A=, B+ etc will allow a total of 1.2A output across the USB ports regardless of whether this setting is configured on not?

XECDesign, 2017-01-10:
I don't know if it is. I asked @pelwell and he checked the firmware source. Instead of checking the option, that part of code now sets the appropriate pin, if it's present.

pelwell, 2017-01-10:
Models B+, A+ and Pi 2 have a GPIO pin to enable the variable limit, but on other models there is no such control. The configuration setting was removed in October.

Romeo-Golf, 2017-01-10:
Cheers, ive found the relevant commit.. and although it doesnt seem clear why this is removed, I doubt there will be many complaints!

https://github.com/raspberrypi/firmware/commit/a021d6e70bf5a710c364d21852be4a8dd4dc423a