RPi-Distro/raspi-config #43: Support disabling/enabling Bluetooth on Raspberry Pi 3

itemir, 2017-01-06:
Raspberry Pi 3 comes with Bluetooth. While it is very nice to have, it also occupies UART and causes custom applications that rely on UART to fail.

It would be nice to have a mechanism to enable/disable Bluetooth, for reasons including but not limited to above (another reason is potential security/privacy).

XECDesign, 2017-01-10:
Thanks, @itemir. @spl237 is taking a look at adding the option. It might be slightly different to your PR.

itemir, 2017-01-10:
Thanks @XECDesign, @spl237. I changed the PR based on the discussions there. Have a look and let me know if it makes sense.

JamesPi123, 2017-04-25:
BT controls are in the system tray.

Nightblade, 2019-04-01:
The system tray is not available to CLI sessions.

I think this would be a good idea.  Probably should add an option to dis/en-able the built-in wifi too.
Perhaps ``/boot/config.txt`` could be leveraged to kill two birds with one stone?
```
dtoverlay=pi3-disable-wifi
dtoverlay=pi3-disable-bt
```

itemir, 2019-04-01:
Closing due to age.

itemir, 2019-04-01:
If you need this capability, check out https://github.com/itemir/rpi_boat_utils/tree/master/uart_control

closed 2019-04-01 by itemir

Nightblade, 2019-04-01:
Thanks for the suggestion, but from a new-user point-of-view, `raspi-config` works out-of-the-box, and is easier to use.  Also, given what it can already do, it does seem like a glaring omission to exclude WIFI/BT *IMHO*.

itemir, 2019-04-01:
> Thanks for the suggestion, but from a new-user point-of-view, `raspi-config` works out-of-the-box, and is easier to use. Also, given what it can already do, it does seem like a glaring omission to exclude WIFI/BT _IMHO_.

Agree with that, that's why I had had the suggestion in the very first place. But after 2 years of inaction, no need to keep this open.