RPi-Distro/raspi-config #46: Enable permanently configuring governor

DavidM42, 2017-02-11:
I made a small script to permanently change the cpu governor and up threshold on the raspi by always changing it at boot with a init.d script. 
Unfornately I am not experienced enough to integrate the functionality myself into this repo, although I think it would be a useful advanced option especially for high power applications like openvpn server or low power like sensors.
 [**Repository with the standalone application code**](https://github.com/DavidM42/rpi-cpu.gov)

DavidM42, 2017-02-11:
I would expand the readme of my repo if it helps or try to make a pull request myself If you could point me in the right direction how the structure of the code works.

XECDesign, 2017-02-23:
I am guessing this is outside of the scope of raspi-config. 
@spl237 ?

spl237, 2017-02-23:
In general we do not add functionality which we do not recommend to raspi-config, and this sort of change to the governor would come under this category.

closed 2017-02-23 by spl237