RPi-Distro/raspi-config #47: Add release version to 'About raspi-config'

craic, 2017-03-03:
Is the version of raspi-config tied to a specific Raspbian release ? _It looks to me like it is not.._

If not, then can we add a version number to the script and have that listed in the 'About raspi-config' section ?

I recently ran into an issue that involved interface settings and had no easy way to telling if the requestor was using the same version as I was.

If you can come up with a current release number then I'll add this myself and do a pull request

thanks

XECDesign, 2017-03-06:
I would suggest using 'apt-cache policy raspi-config' to figure out what version they are running.

closed 2017-04-03 by XECDesign

patricktokeeffe, 2017-08-16:
I would also like to see this revision. At risk of overgeneralizing, the About box usually gives a description, some authorship and version info, as well as project contact info. 

The relevant blurb is [here](https://github.com/RPi-Distro/raspi-config/blob/d2ee57cb1c83113f1381c123f6f84cb0dc26bfc7/raspi-config#L81). Inserting just the version string wouldn't be too difficult.