RPi-Distro/raspi-config #53: There was an error running option 8 Update

RUDI-TERROR, 2017-04-21:
Hi,

when I try to update my raspi-config I get folowing error:

 **There was an error running option 8 Update**

When I open the Option 6 -> Overclock I get following error:

**This Pi cannot be overclocked.**
Then
**There was an error running option 6 Overclock**

When I try to activate the Boot to Desktop and activate RealVNS server, nothing happens, after reboot. 

I Try:
apt-get remove --purge raspi-config && apt-get install raspi-config

But the error ist the same. I use a raspi 3 wirh the latest raspbian firmware. 

Can anyone help me?

Thx! Rudi

XECDesign, 2017-04-24:
As the error message says, the overclock option is not suitable for the pi model you're using.

Regarding RealVNC, what do you mean by 'nothing happens'? What do you expect to happen? How have you determined that it's not happening? What image are you using? Is it up to date?

RUDI-TERROR, 2017-04-25:
Yes, I use a actual firmware:
root@mapi:/# uname -a
Linux mapi 4.9.23-v7+ #989 SMP Wed Apr 19 15:29:25 BST 2017 armv7l GNU/Linux

I do follow steps:
root@mapi:/# raspi-config
-> 3 Boot Options
-> B1 Desktop / CLI
-> B3 Desktop
-> Finish
After reboot the pi doesn't boot to Desktop, the pi boots to console. When I use the command startx the desktop start.

Is there an error log for raspi-config?

XECDesign, 2017-04-25:
This doesn't sound like a raspi-config issue.

What's the distro? What's the image?

RUDI-TERROR, 2017-04-26:
I use raspbian..

XECDesign, 2017-04-26:
Lite or with PIXEL?

RUDI-TERROR, 2017-04-26:
Pixel

XECDesign, 2017-04-26:
What do you get when you run `systemctl get-default`?
What happens when you run 'sudo systemctl start lightdm'?

RUDI-TERROR, 2017-04-26:
root@mapi:/home/pi# systemctl get-default
graphical.target
root@mapi:/home/pi#

After I run sudo systemctl start lightdm and closy kodi i see Pixel Desktop

XECDesign, 2017-04-28:
That tells us that raspi-config did what it's supposed to. Maybe the full output of 'sudo journalctl' after booting would say why lightdm isn't starting automatically.

closed 2017-07-11 by XECDesign