RPi-Distro/raspi-config #55: Command line options

walchko, 2017-04-23:
Could command line options be added so this could be run from a script? At times I setup a bunch of these and it gets tedious going through all of the menus. Or is that not possible (or maybe better words would be "too difficult")?

XECDesign, 2017-04-24:
That's already an option. https://github.com/raspberrypi-ui/rc_gui/blob/master/src/rc_gui.c#L23

walchko, 2017-04-24:
wow, thanks! I googled around and never found this ... maybe I didn't try hard enough or pick the right keywords. I suggest this gets added to some documentation somewhere (maybe a readme.md in the repo). Thanks again!

closed 2017-04-24 by walchko

XECDesign, 2017-04-24:
Pull requests for documentation are always welcome. raspi-config documentation has been in need of an update for a while, I think.

walchko, 2017-04-24:
Sure, give me some time (I have to play with what you showed me) and I will do something.