RPi-Distro/raspi-config #61: Stretch raspi-config tiny problem

pcollinson, 2017-08-31:
The new version of raspi-config for stretch is a serious improvement if you don't want to be the 'pi' user. The change to allow it to set things up for the sudo'd user is great. There are, however, some holes.

On a vanilla system, I changed my Desktop default user to me - rebooted and logged in.. Great - the default desktop user was me. However, the default underlying tty account was opened for 'pi'. The work around was to reboot as cli user with 'me' as the default user, and then reboot again as the desktop user. The extra account opened by systemd disappears.

Looks as if that when you change the default user - it needs to change both the lightdm X setup and the terminal code too.

This is not a showstopper - but I thought it could be reported. I think I put this in the wrong place initially. Hopefully this is right now.

XECDesign, 2020-01-13:
Sorry for the very late response, but I'm not sure I understand the problem. Could you please give instructions on how to reproduce the issue, what the expected behaviour is and what the actual behaviour is.

I don't see how you changed the default user, which raspi-config options you used or what you mean by 'underlying tty account'.

pcollinson, 2020-01-13:
This problem has gone away in subsequent releases of raspi-config..

closed 2020-01-13 by pcollinson