RPi-Distro/raspi-config #62: Connect to WiFi?

nickoala, 2017-09-06:
Has anyone considered adding a "Connect to WiFi" option in raspi-config? One that asks for SSID and passkey and insert the appropriate contents into `wpa_supplicant.conf`?

For Linux newbies, generating the network block and inserting that into the file represents a lot of hassle. I think having an easier way to do that would make the command line feel a little less formidable to beginners.

I can create a PR if maintainers have no objection :smile:

nielsnl68, 2017-09-29:
I love this option,  too

davidplappert, 2017-10-02:
I would like this too. I have a Pi Zero W. A couple features I had in mind.

- Make it non-interactive and interactive
- show a list of detected networks when setting SSID on interactive mode(`sudo iwlist wlan0 scan `)
- only make the wireless config option show on Pis that have Wireless (raspi-config already has functions for detecting pi versions)
- allow for public wifi (no password)
- Enable more than 1 network to be entered (home/school/work)

Re: I just read the Pi doc and generated feature requests from that along with my personal experiences.

https://www.raspberrypi.org/documentation/configuration/wireless/wireless-cli.md

Much thanks. Let me know if you want any help.

sbf-, 2017-10-02:
> only make the wireless config option show on Pis that have Wireless

Every RPi can have wireless (with WiFi dongle) so it should not be limited to Zero W or 3
Also, on Stretch, wlan0 doesn't exist - it's something with MACaddress (eg wlx00c140123456) - unless you specify net.ifnames=0 in /boot/cmdline.txt

davidplappert, 2017-10-03:
> Also, on Stretch, wlan0 doesn't exist

I have a Pi Zero W, downloaded Stretch from Raspberry Pi's site, and wlan0 worked out of the box with the documentation: https://www.raspberrypi.org/documentation/configuration/wireless/wireless-cli.md

XECDesign, 2017-10-03:
I would love the option, but it's not as straight forward as it might seem. The easy solution of "Enter SSID" "Enter passphrase" will work. If you want a nice menu to select the SSID and support profiles and configure static IP for each interface and do all the other things people will ask for, you'll end up with something much larger than what should be in raspi-config.

If the easy solution is good enough, then it's not much of an improvement over  `wpa_passphrase SSID PASSPHRASE >> /etc/wpa_supplicant/wpa_supplicant.conf`

davidplappert, 2017-10-03:
> I would love the option, but it's not as straight forward as it might seem

I would be on board even with simple ssid/psk fields.

sbf-, 2017-10-03:
@davidplappert So it seems they changed it back to wlan0... I have an image of 16-AUG-2017

nickoala, 2017-10-04:
> I would love the option, but it's not as straight forward as it might seem ...

Good that someone brought this up. It's one of my "worries" too ... That being said, my intention was never to duplicate the whole range of things one can do by editing the config files by hand. 

My intention was to give *beginners* an easy way to connect to WiFi from the command line (without seeing the PIXEL desktop). A simple "Enter SSID/Enter passphrase" (as suggested by @XECDesign and echoed by @davidplappert) was what I originally had in mind.

It may not be a huge improvement over `wpa_passphrase ssid passphrase | sudo tee -a /etc/wpa_supplicant/wpa_supplicant.conf`, but trust me, that can make a huge difference to *beginners* who have never used Linux before. It's OK to ask them to type `ifconfig` or `sudo iwlist wlan0 scan`. But anything involving a pipe and changing a root-owned file and you have to explain to them what that series of characters actually means ...... it's hopeless. And it makes the command line too intimidating.

Here is what I will do:

1. Add an option which leads to: **Enter SSID / Enter passphrase**
2. If passphrase is empty, assume open public network
3. **Replace** the network block in `wpa_supplicant.conf` if the supplied SSID is present in the file. Otherwise, **append** the network block.

Anything more complicated, it's time to learn the Linux way and edit the files by hand!!!

XECDesign, 2017-10-04:
Sounds good to me.

@spl237, any objection?

spl237, 2017-10-04:
Sounds good to me too. Can I suggest it is included in the menu structure by the following:

1) Create an item in the top level menu called "Network Options" and included it at number 2 to replace "Hostname"
2) Create a submenu for "Network Options" with "Hostname" moved into it, along with the new "Enter SSID / passphrase" option.

That is then nicely consistent with the rest of the top level menu.

XECDesign, 2017-10-04:
Then I guess predictable interface names option should be moved in there too? 'Wait for network' is another candidate, but it might be more boot-related than network-related.

spl237, 2017-10-04:
Yes, that's a good place for predictable interface names. I'd leave wait for network under boot options.

nickoala, 2017-10-04:
OK. Number 2 becomes "Network Options", which includes under it:
- Hostname
- Enter SSID / passphrase
- Predictable interface names

Expect a PR within a few days :blush:

nielsnl68, 2017-10-04:
Very good. I have being waiting for this for a long time.

I have Just one related wish; How about going one extra step and allow to
set dhcp/fixed ip within the raspi-config ?

XECDesign, 2017-10-04:
Let's stick with one thing at a time for now, but yes, static IP setup is something worth discussing.

nickoala, 2017-10-11:
Just created a PR #64

It's a little more sophisticated than originally envisioned. It scans for hotspots before asking users to enter an SSID manually. More user-friendly, but at the same time introduces some challenges. Please see the PR for details.

nickoala, 2017-10-23:
Attempt Number 2: PR #68 

Scrap hotspot menu. Stick to manually entering SSID and passphrase :smile:

XECDesign, 2017-10-27:
Fixed by https://github.com/RPi-Distro/raspi-config/commit/1e0503af817e0aee6a933657901021378c776129

closed 2017-10-27 by XECDesign