RPi-Distro/raspi-config #65: raspi-config behaves strangely when one tries to desactivate VNC

ussmarc, 2017-10-12:
I've noticed two small bugs in the version of **raspi-config** that ships with the **2017-09-07** version of Raspbian. To reproduce the first one, install a fresh copy of 2017-09-07-raspbian-stretch-lite.img on a card and boot the Pi. Launch raspi-config, and try to explicitly desactivate VNC : raspi-config replies that VNC is **enabled**, which is the exact opposite of what one wanted to do.

To reproduce the second bug, try to update raspi-config to the latest available version, which is a little more recent than the one included in the 2017-09-07 image, in case the first bug would have be corrected, and try to desactivate VNC again : raspi-config replies that VNC is... <nothing at all> (an empty string is displayed instead of the word "desactivated").

XECDesign, 2017-10-23:
You're not the first person to report the issue, so I believe there's something to it - https://github.com/RPi-Distro/raspi-config/issues/57. However, running the latest version of raspi-config, I am seeing the correct and expected behaviour. 

Someone who is experiencing the issue would need to dig  a little deeper. What does `systemctl status vncserver-x11-serviced` return?

XECDesign, 2017-10-30:
No response.

closed 2017-10-30 by XECDesign