RPi-Distro/raspi-config #74: typo in VNC-Setting (P3)

ftg5z, 2017-12-25:
if I set the option to "no" it results in message  <pre>"The VNC Server is enabled"</pre> or <pre>"The VNC Server is                   "</pre>
But apt is not executed if I click on "no", so it's not a technical problem.

Terminal says:
Unit vncserver-x11-serviced.service could not be found.
Failed to disable unit: No such file or directory

patricktokeeffe, 2018-02-15:
Confirming present on clean Raspbian Lite Nov 29, 2017 install. No harm, no foul (tho prob does confuse some users). 

xref: #57, #65

shatteredsword, 2018-05-24:
raspbian lite isn't running an x server, so vnc won't work anyway. it seems like the correct way to fix this issue would take a ton of work as it would require checking to see if the x server is running correctly etc. and then just remove the menu option if it's not.

commit cabd4289 (2020-01-13): Fix VNC options

Fixes #74
closed 2020-01-13 by XECDesign