RPi-Distro/raspi-config #78: Enabling GL driver has no effect on renderer used by X, glxinfo,...

jjakob, 2018-03-22:
On stretch, enabling the GL driver via raspi-config has no effect on the renderer used by X, glxinfo, etc.
Rendering as reported by glxinfo and glxgears stays via llvmpipe, resulting in no change of graphics performance to users.
After disabling /usr/share/X11/xorg.conf.d/99-fbturbo.conf (by renaming it) glxinfo reports proper VC4 rendering and glxgears gives ~1000fps.
Perhaps raspi-config should do this automatically when enabling/disabling the GL driver. Or is there a different way by which VC4 should override fbturbo?

```
~# apt-cache policy raspi-config 
raspi-config:
  Installed: 20180228
  Candidate: 20180228
  Version table:
 *** 20180228 500
        500 http://archive.raspberrypi.org/debian stretch/main armhf Packages
        100 /var/lib/dpkg/status

```
```
~# cat /etc/debian_version
9.4
```

glxgears -info after renaming 99-fbturbo.conf:
```
libGL error: MESA-LOADER: failed to retrieve device information
MESA-LOADER: failed to retrieve device information
MESA-LOADER: failed to retrieve device information
Running synchronized to the vertical refresh.  The framerate should be
approximately the same as the monitor refresh rate.
GL_RENDERER   = Gallium 0.4 on VC4 V3D 2.1
GL_VERSION    = 2.1 Mesa 13.0.6
GL_VENDOR     = Broadcom

```

jjakob, 2018-03-23:
I should add that this is only if xserver-xorg-video-fbturbo is installed.

XECDesign, 2018-03-23:
It seems to rely on having the `gldriver-test` package installed.

jjakob, 2018-03-23:
Interesting, gldriver-test provides usr/lib/systemd/scripts/gldriver_test.sh which checks if the v3d device is present and operating and removes/recreates 99-fbturbo.conf as necessary.
Shouldn't this be a dependency of raspi-config? Otherwise people have no idea why it's needed.
Also noticing that the script has a template file hardcoded into it so any changes to 99-fbturbo.conf will be lost when enabling/disabling the GL driver.

commit 4d1dc2e2 (2020-01-13): Install required packages when changing GL settings

Fixes #78
closed 2020-01-13 by XECDesign