RPi-Distro/raspi-config #80: Alpine Linux support

yarons, 2018-06-28:
Hey guys, how can I help with adjusting the scripts to Alpine Linux?
There is another set of scripts called "Alpine setup scripts" which contains most of the basics of the operating system:
https://wiki.alpinelinux.org/wiki/Alpine_setup_scripts

I'm still thinking about maybe adjusting this script to be included in this set of scripts.

XECDesign, 2018-06-28:
raspi-config is intended to be used with Raspbian, so it won't be adjusted to suit distros we don't support. I think if it was to be included with other distros, it's better done as a fork. Otherwise we'd have a switch statement for each option, only one of which would ever get tested.

sbf-, 2018-06-28:
alpi-config might be a good name ;-)

yarons, 2018-06-28:
@XECDesign I'm confused, the scripts can easily detect the OS and act accordingly, it can be merged if needed.

XECDesign, 2018-06-28:
Yes, that's what I meant by the switch statement.

Who is going to maintain that and make sure all the options are tested on all the distros? Who's going to find out how to something one every distro when we add an option? I'm not comfortable with the maintenance burden.

closed 2018-07-03 by XECDesign