RPi-Distro/raspi-config #81: Auto login settings are getting reset.

joolswills, 2018-07-23:
Could be argued this is not a raspi-config issue, but I wanted to give a heads up on it as it affects raspi-config.

I opened an issue against Raspbian here some months ago -https://bugs.launchpad.net/raspbian/+bug/1755811 but had no feedback from any developers.

When there are systemd package updates, any config to automatically log in is removed. I assume the fix for this would be for Raspbian to ship with a custom systemd package that doesn't have this behaviour.

RetroPie installed on Raspbian uses raspi-config to automatically log in, and we are getting a lot of reports about this, so I was hoping to get someone's attention.

Perhaps there is another way to automatically log in, that the systemd package won't remove on upgrade ?

cmitu, 2019-04-09:
This seems have been solved by the latest upgrade to `raspi-config`. I noticed `systemd` was upgraded today (_systemd:232-25+deb9u11_, from the Debian security updates) and the auto-login settings have been preserved.

XECDesign, 2019-04-09:
Probably thanks to this commit, which handles autologin much better than we were before: https://github.com/RPi-Distro/raspi-config/commit/d33487cbeac581d2ce91d8372f5feb1e2b5b7e3a

purpleidea, 2019-12-16:
I just downloaded the latest release: 4.5.1 and after an `apt update && apt upgrade` I can confirm this issue occurred. I then did the RetroPieSetup upgrade, and after that long thing finished, the issue was still present. HTH

XECDesign, 2020-01-09:
How can I reproduce the issue? I've seen people mention it in the context of RetroPie, so maybe that's where the problem is.

joolswills, 2020-01-09:
I thought this was resolved now - 4.5.1 was released 5 months ago so the raspi-config included may not have included the changes to raspi-config that fixed this.

RetroPie-Setup just calls the raspi-config script btw - it doesn't do anything differently. It used to occur when setting to auto login, then updating systemd to a new version would reset it - there's more information in the launchpad bug on the first post.

joolswills, 2020-01-09:
I think this is fixed on Buster - RetroPie 4.5.1 uses Stretch so may still have the issue.

XECDesign, 2020-01-09:
Alright, thanks Jools. That makes sense. Closing the issue since it's fixed in Buster.

closed 2020-01-09 by XECDesign