RPi-Distro/raspi-config #93: Cannot resize partion

Muffo, 2019-04-11:
My root partition is full:

```
pi@hassbian:~ $ df -h
Filesystem      Size  Used Avail Use% Mounted on
/dev/root       1.8G  1.7G   44M  98% /
devtmpfs        460M     0  460M   0% /dev
tmpfs           464M     0  464M   0% /dev/shm
tmpfs           464M  6.2M  458M   2% /run
tmpfs           5.0M  4.0K  5.0M   1% /run/lock
tmpfs           464M     0  464M   0% /sys/fs/cgroup
/dev/mmcblk0p1   43M   22M   22M  51% /boot
tmpfs            93M     0   93M   0% /run/user/1000
```

and I am trying to resize it with raspi-config, but after rebooting the RPi nothing happens.
Here is a screenshot of the output that briefly appear when I run the command:

[image]

Maybe it is because of the 2 error messages?
I have already tried to reboot the device multiple times, but I get always the same result.

Muffo, 2019-04-11:
It turns out I needed to run this command after reboot:

sudo resize2fs /dev/mmcblk0p2

XECDesign, 2019-04-12:
Just tested it and it worked fine.

If you're able to go back, after you select the resize option and reboot for the first time, what is the output of `sudo journalctl -u resize2fs_once`?

Muffo, 2019-04-13:
After running the command above I was able to resize the partition so unfortunately I can't gather more info on the problem.
Feel free to close the ticket since I understand that there aren't enough info for debugging.

XECDesign, 2019-04-14:
Alright. I'll close this for now. Thank you.

closed 2019-04-14 by XECDesign