RPi-Distro/raspi-config #95: expand partition question

dongdixidi, 2019-06-11:
I noticed that the methods to expand partition are diffirent between raspi-config and init_resize.sh, the raspi-config is use fdisk command; and the init_resize.sh is use parted command.I wanna known what is diffirent between thoose two methods.
 And I found that on 2018-06-27-raspbian-stretch-lite.img, when I did some changes ,then I backup and restore it to a new image by tar, the init_resize.sh seems not used; but raspi-config can expand partition.
I known that may not be an issue, but can you help me?

XECDesign, 2019-06-11:
No difference, they're just two different tools which share functionality. Off the top of my head, I can't think of any reason to prefer one over the other.

> I known that may not be an issue, but can you help me?

Help with what? For general support, the forum is your best bet.

https://www.raspberrypi.org/forums/index.php

dongdixidi, 2019-06-13:
Thanks for your reply , I found the way to expand partion on the first boot in [https://www.raspberrypi.org/forums/viewtopic.php?f=91&t=241153&p=1471365&hilit=init_resize#p1471365](url).
It seems to need add resize2fs_once script. And after I do it, it's works

closed 2019-06-13 by dongdixidi