RPi-Distro/raspi-config #97: 2019-06-20 buster lite, generic keyboard selections missing

georgejhunt, 2019-07-12:
Steps to re-create:
1. validate shasum of downloaded zip file (avoid "latest"  which is inexact):
2. https://downloads.raspberrypi.org/raspbian_lite/images/raspbian_lite-2019-06-24/2019-06-20-raspbian-buster-lite.zip
3. Copy to a second SD card (I've experienced dd to fail silently on defective SD cards).
4. boot
5. log in as pi/raspberry
6. sudo raspi-config
7. localization/keyboard/
8. notice that generic-101 is not available. Also there are issues (I think possibly related to fonts) showing square black dots where characters should be -- more at the bottom of the list.
(I have used this url, screen capture #3 for a refresher of what I have seen on earlier releases -- 
https://thepihut.com/blogs/raspberry-pi-tutorials/25556740-changing-the-raspberry-pi-keyboard-layout)
This problem does not occur on the desktop version. This problem does not go away with "apt update","apt upgrade".
About a week ago, I remember encountering this problem. I vaguely remember changing the locale, and trying lots of other things like apt upgrade, in some order. I got around the problem once, but I cannot replicate.
Our "internet-in-a-box" team uses a one line installer strategy:
#                 curl d.iiab.io/install.txt | sudo bash
-- which is pretty difficult if the keyboard does not offer (or I don't know how to find) a '|' pipe symbol!

XECDesign, 2019-07-12:
I believe this is an upstream Debian issue that should be fixed in the next point release.

georgejhunt, 2019-07-14:
Thanks for the response, and the understanding of time frame for correction

XECDesign, 2020-01-09:
AFAIK, this has been fixed in Buster. Please re-open if I'm wrong.

closed 2020-01-09 by XECDesign