RPi-Distro/raspi-config #101: raspi-config makes assumption it is being run through sudo

maxnet, 2019-10-16:
raspi-config seems to be making the assumption it is being run through sudo, as several functions seems to assume $SUDO_USER is set.

Keep in mind that some users may start it as root in other ways though.
E.g. through the piserver console
Or they may have set a password for the root user and log in directly through that, or indirectly through "su" without the "do".

May want to print a warning if SUDO_USER is empty, and an option is called that does expect it to be set.

ZanderBrown, 2019-10-16:
Or run through pkexec

mattvw, 2019-10-16:
Agreed...would be really beneficial to have this check be in place.

I've run into issues in the past when there was no notice that it wasn't setting things correctly since it wasn't being run through sudo, and then it took me a lot of time to figure out what the issues were and what had to be done to make it set correctly. Especially with the GUI/console auto-login settings.

PR #104 (2020-01-13): Handle cases where SUDO_USER is unset
Fixes #101, closes #76
closed 2020-01-13 by XECDesign