RPi-Distro/raspi-config #102: Why not load overlay module from /lib/modules ?

maxnet, 2019-10-16:
Your new overlayfs functionality seems to stick the overlay kernel module in the initramfs archive, making the initramfs kernel version specific.

Since you do have everything else needed to access storage and mount the ext4 file system into the kernel, I was wondering why not just load the module off the ext4 file system instead?

Along the lines of: (not tested)

```
	# Mount read-only root to /lower
	if [ "${FSTYPE}" != "unknown" ]; then
		mount -r -t ${FSTYPE} ${ROOTFLAGS} ${ROOT} /lower
	else
		mount -r ${ROOTFLAGS} ${ROOT} /lower
	fi

	if ! grep -q "overlay" /proc/filesystems; then
		# overlayfs not in kernel, load extra module from ext4 file system
		insmod /lower/lib/modules/`uname -r`/kernel/fs/overlayfs/overlay.ko
	fi
```

XECDesign, 2020-01-13:
Out of curiosity, is there a particular use case that raises this issue?

In some ways /boot files (kernel and initramfs) are more closely tied to each other than they are to rootfs. If you can get in a state where your kernel does not match your initramfs, you can also end up with initramfs not matching modules/. I don't know which is more likely.

maxnet, 2020-01-13:
>Out of curiosity, is there a particular use case that raises this issue?

Main concern is that people are known to replace their kernel in multiple non-standard ways (e.g. by running rpi-update) or by a custom kernel that was cross-compiled on a different computer, and not all those ways may call the update-initramfs stuff to generate a new initramfs with the matching modules.

>If you can get in a state where your kernel does not match your initramfs, you can also end up with initramfs not matching modules/

Well, if there is no matching modules directory, you are going to have a pretty useless system anyway.
Given that Raspbian boots to X by default, X requires evdev for keyboard/mouse to function, but it is a module in your kernel configuration...
(Something that should also be fixed in my opinion, but that is a different discussion).

XECDesign, 2020-01-13:
Thanks for clarifying

XECDesign, 2020-01-13:
Thank you, that makes sense. I'll take a look.

XECDesign, 2020-01-17:
#105 seems to work. Any objections to that approach?

maxnet, 2020-01-17:
Should be fine as long as the script continues if both fail, and does not error out.
The circumstances in which both fail would be if it is a custom kernel that has overlayfs statically compiled into the kernel instead of as module.

XECDesign, 2020-01-17:
Thanks. It seems to continue if I put a `false` in the script.

PR #105 (2020-01-17): OverlayFS - try loading module from rootfs
Fixes #102
commit de8a8dad (2020-01-17): OverlayFS - try loading module from rootfs (#105)

Fixes #102
closed 2020-01-17 by XECDesign