RPi-Distro/raspi-config #108: Overlay-FS: what happens when disabling and enabling later again?

bouwew, 2020-02-21:
Hi, many thanks for adding this function!

But, now I'm using it and while developing have been switching it "on", "off" and back "on".
I have the impression that after switch it back on, the older files that are still present on lower are not being updated by the newer files when the overlay-fs is enabled again. So, shouldn't lower be cleared and/or unmounted when overlay-fs is disabled?

XECDesign, 2020-02-21:
are you rebooting after toggling the setting?

bouwew, 2020-02-21:
Yes, always.
Today I made a change and it stayed after putting it back to RO mode. So, scratching my head...

XECDesign, 2020-02-21:
Not quite sure what you mean.

You write some files. You turn on the overlay. From that point, the files will remain there, but no other changes will survive reboot.

You turn off the overlay and you can make permanent changes in addition to the ones you made initially.

That's how it's meant to work. How does what you're seeing differ?