RPi-Distro/raspi-config #111: Restart of cron missing when timezone is updated

framps, 2020-04-13:
I just got an issue for [raspiBackup ](https://github.com/framps/raspiBackup/issues/217) and finally it turned out the root cause for the issue is located in raspi-config. The user used raspi-config to change the timezone and detected cron starts tasks at an incorrect time. According [this page](https://serverfault.com/questions/231220/why-are-cronjobs-running-at-the-wrong-time) a restart of cron has to be done whenever the timezone was changed.

It's not a serious issue because sooner or later the Raspberry will be restarted but it's a nasty issue in particular for beginners. I suggest to restart cron when the timezone is changed in raspi-config.

XECDesign, 2020-04-14:
It seems like this wouldn't be a problem unique to cron and raspi-config should prompt the user to restart the system instead.

framps, 2020-04-14:
I don't think a user of raspi-config should be prompted for a restart all the time. `systemctl restart cron` will restart cron very quickly.

XECDesign, 2020-04-14:
Right, but what makes cron special to any other software that may not cope with a timezone change?

Since we don't know what other software may be affected, it seems like a restart is the right solution, as suggested on the page you linked. Otherwise we'd be playing whackamole with all the special cases that may need to be handled.

On the other hand, I don't have a strong preference either. Given that this is the first time anybody has noticed this, it's unlikely to come up again with other software, so just handling cron makes sense as well.

So maybe we should restart cron and then if the problem is reported again with other software, we can prompt for a reboot rather than adding special cases.