RPi-Distro/raspi-config #116: raspi-config does not prompt username if is ran under "su"

slrslr, 2020-05-05:
Hello, i am linking to a discussion topic where user was confused due to raspi-config not asked for which user to apply actions.

[Here](https://www.raspberrypi.org/forums/viewtopic.php?p=1655654#p1655654) is the issue described

When raspi-config is ran under "su" it not asked, but when ran directly under super user logged in, it asked.

Please consider prompting for the username always.

PixelOFC, 2020-06-04:
run `sudo raspi-config`, when you ran it using `su` you automatically enter in `root` user.

XECDesign, 2020-06-05:
It tries to figure out what user you're logged in as and will only fall back to asking if it can't. That was added because people were logging in as root and then breaking things like auto-login.

It's a bit of a confusing question to ask a beginner, so I'd like to avoid it as much as possible. I understand that the opposite can be confusing too, but most people won't be using multiple accounts and the menu does mention the user name in the descriptions.