RPi-Distro/raspi-config #117: Kernel panic on first boot

edurenye, 2020-05-09:
Booting the RPi 3 B+ with the last Raspbian set up using pi-gen got me the following error:

```
---[ end Kernel panic - not syncing: Requested init /usr/lib/raspi-config/init_resize.sh failed (error -13). ]---
```

RaspberryPiFan, 2020-05-21:
This means the init script has the permissions set wrongly (e**x**ecute permission missing)

edurenye, 2020-05-21:
Is the pi-gen that sets those permissions I haven't changed them.

Actually it just works fine if before I use the dd command to move the image to the SD card I format the SD card.
If first I do not format it then this error appears, it's kind of strange. So now I always just format it before the dd command.