RPi-Distro/raspi-config #118: raspi-config assumes configuration files are in /boot

Monofin, 2020-05-15:
Ubuntu aarch64 places the Pi's firmware and config files in /boot/firmware by default, and raspi-config will fail as a result when trying to update the configuration or overlays.

XECDesign, 2020-05-15:
We don't support third party distros

Monofin, 2020-05-15:
(Assign this to me if you like, I have a patch that parameterises and checks for Ubuntu?)

XECDesign, 2020-05-15:
Are the config files auto-generated? I've seen some setups where that happens, in which case raspi-config changes would be overwritten when they get re-generated.

How much complexity does the patch add?

My concerns is that gentoo, arch and all the others would need to be supported too, so I would rather this sort of patch was applied downstream.

Monofin, 2020-05-15:
NP. I'll make a few more tests for auto generation, it may be a dpkg thing. Management of that many distro would become a headache, I understand if you don't want to open the floodgates!

dl6dbh, 2020-12-05:
ich habe ein debian 10 buster 64 bit laufen 
kann ich das tool darauf laufen lassen ?

closed 2020-12-07 by XECDesign