RPi-Distro/raspi-config #120: Work of raspi-config in chroot

urpylka, 2020-07-03:
Hello, I am making **preconfigured RPi images**.
For that I use `chroot` & `qemu-static` in [the docker image](https://github.com/urpylka/img-tool) (for [translate to target architecture](https://github.com/urpylka/img-tool/blob/master/src/qemu-wrapper.c)).

Now, I've made https://github.com/CopterExpress/clover/blob/master/builder/assets/hardware_setup.sh#L42 For ex:

```bash
/usr/bin/raspi-config nonint do_i2c 0
```

It's work at the first boot OS. But when I try to execute this in `chroot`, it causes errors, like:

```bash
mount: /config/device-tree: mount point does not exist.
* Failed to mount configfs - 2
modprobe: ERROR: ../libkmod/libkmod.c:586 kmod_search_moddep() could not open moddep file '/lib/modules/4.19.76-linuxkit/modules.dep.bin'
modprobe: FATAL: Module i2c-dev not found in directory /lib/modules/4.19.76-linuxkit
```

And I want to know could I use `raspi-config` in `chroot` to configure hardware?
Does it possible to make it works in `chroot`?

eudoxos, 2023-02-27:
As long as it needs to access hardware, including the device tree, it can't work in chroot. I am using `proot` for pre-configured images, that works well, but wrote  all the steps normally done through `raspi-config` in normal bash commands (e.g. `usermod`, `systemctl enable` etc).

I would like to have some more documentation on both `raspi-config nonint` and also documentation on which part of the `nonint` modifications can be done in chroot (for root), which additionally need `/boot` access (has to be bind-mounted in the chroot), and which must be run on hardware.

XECDesign, 2023-02-27:
https://github.com/raspberrypi/documentation/issues/2454

nonint functionality exists only for internal use and can change without notice, so any detailed documentation would be out of date very quickly and do more harm than good.

I would like to be able to spend time and document everything properly and make sure that all changes happen in tandem with documentation changes, but that's just not possible without time being taken away from other more important things.

closed 2023-02-27 by XECDesign