RPi-Distro/raspi-config #121: nonint mode should verify that the commands being run are valid

TonyLHansen, 2020-07-05:
At a minimum, nonint() should check that $1 begins with "do_" before invoking it:

```
nonint()
{
    case "$1" in
        do_* ) "$@" ;;
        * ) printf "Invalid nonint command '%s'\n" "$1"
    esac
}
```

TerryE, 2023-02-01:
@TonyLHansen, there are a whole bunch of functions not starting with `do_` that are perfectly valid and used, for example those starting with  `is_`, `get_` ,  `enable_`, ``disable_`, etc.  Also `do_anything` would pass your test but this would still error.  `raspi-config` uses `dash` as its shell and this interpreter doesn't have introspection, so any non-trivial `nonint` validation would require pretty much doubling the codebase.  Scripted use of this tool can be extremely useful but there is an implicit assumption that the scripture can understand shell scripting and does any necessary validation in the calling scrippt.

XECDesign, 2023-02-06:
nonint use is unsupported, undocumented and subject to change without any notification. It's only intended to be used in a tight set of circumstances - rc_gui, pi-gen and so on. If somebody else is using it, we assume they know what they're doing.

closed 2023-02-06 by XECDesign