RPi-Distro/raspi-config #122: Default gpu_mem is now 76MB, not 64MB

andrum99, 2020-07-06:
raspi-config thinks the default setting for `gpu_mem` is 64MB - it's not. If there is no `gpu_mem` directive specified in `config.txt`, the firmware now defaults to allocating 76MB to the GPU. `raspi-config` should be modified accordingly.

andrum99, 2020-07-06:
It appears the Pi 1 (and probably Zero) have the old default, i.e. 64MB.

lurch, 2020-07-07:
Just for reference, @popcornmix said [here](https://github.com/raspberrypi/documentation/pull/1610#issuecomment-654778825) that:
"default gpu_mem is 76 for 1GB (or greater) and 64 for less."

andrum99, 2020-07-07:
> Just for reference, @popcornmix said [here](https://github.com/raspberrypi/documentation/pull/1610#issuecomment-654778825) that:
> "default gpu_mem is 76 for 1GB (or greater) and 64 for less."

Correct. My wording transposes that onto the models of Pi. Only Pi 1 and Zero have less than 1GB.