RPi-Distro/raspi-config #129: Docker fails to start when OverlayFS is active

rigon, 2020-08-17:
Docker Daemon fails to start when OverlayFS is active.
It stops with the error:

    Error starting daemon: rename /var/lib/docker/runtimes /var/lib/docker/runtimes-old: invalid cross-device link

Here is the full log:

```
INFO[2020-08-17T20:42:47.286873850+01:00] libcontainerd: started new docker-containerd process  pid=1356
INFO[2020-08-17T20:42:47.287767257+01:00] parsed scheme: "unix"                         module=grpc
INFO[2020-08-17T20:42:47.287942308+01:00] scheme "unix" not registered, fallback to default scheme  module=grpc
INFO[2020-08-17T20:42:47.288309985+01:00] ccResolverWrapper: sending new addresses to cc: [{unix:///var/run/docker/containerd/containerd.sock 0  <nil>}]  module=grpc
INFO[2020-08-17T20:42:47.288545684+01:00] ClientConn switching balancer to "pick_first"  module=grpc
INFO[2020-08-17T20:42:47.288767475+01:00] pickfirstBalancer: HandleSubConnStateChange: 0x40007c6850, CONNECTING  module=grpc
INFO[2020-08-17T20:42:47.325692579+01:00] starting containerd                           revision=9754871865f7fe2f4e74d43e2fc7ccd237edcbce version=18.09.1
INFO[2020-08-17T20:42:47.326791185+01:00] loading plugin "io.containerd.content.v1.content"...  type=io.containerd.content.v1
INFO[2020-08-17T20:42:47.326987977+01:00] loading plugin "io.containerd.snapshotter.v1.btrfs"...  type=io.containerd.snapshotter.v1
WARN[2020-08-17T20:42:47.327471614+01:00] failed to load plugin io.containerd.snapshotter.v1.btrfs  error="path /var/lib/docker/containerd/daemon/io.containerd.snapshotter.v1.btrfs must be a btrfs filesystem to be used with the btrfs snapshotter"
INFO[2020-08-17T20:42:47.327646092+01:00] loading plugin "io.containerd.snapshotter.v1.aufs"...  type=io.containerd.snapshotter.v1
WARN[2020-08-17T20:42:47.333669813+01:00] failed to load plugin io.containerd.snapshotter.v1.aufs  error="modprobe aufs failed: \"modprobe: FATAL: Module aufs not found in directory /lib/modules/5.4.51-v8+\\n\": exit status 1"
INFO[2020-08-17T20:42:47.333903197+01:00] loading plugin "io.containerd.snapshotter.v1.native"...  type=io.containerd.snapshotter.v1
INFO[2020-08-17T20:42:47.334076304+01:00] loading plugin "io.containerd.snapshotter.v1.overlayfs"...  type=io.containerd.snapshotter.v1
INFO[2020-08-17T20:42:47.334371297+01:00] loading plugin "io.containerd.snapshotter.v1.zfs"...  type=io.containerd.snapshotter.v1
WARN[2020-08-17T20:42:47.334917434+01:00] failed to load plugin io.containerd.snapshotter.v1.zfs  error="path /var/lib/docker/containerd/daemon/io.containerd.snapshotter.v1.zfs must be a zfs filesystem to be used with the zfs snapshotter"
INFO[2020-08-17T20:42:47.335049894+01:00] loading plugin "io.containerd.metadata.v1.bolt"...  type=io.containerd.metadata.v1
WARN[2020-08-17T20:42:47.335166539+01:00] could not use snapshotter btrfs in metadata plugin  error="path /var/lib/docker/containerd/daemon/io.containerd.snapshotter.v1.btrfs must be a btrfs filesystem to be used with the btrfs snapshotter"
WARN[2020-08-17T20:42:47.335267963+01:00] could not use snapshotter aufs in metadata plugin  error="modprobe aufs failed: \"modprobe: FATAL: Module aufs not found in directory /lib/modules/5.4.51-v8+\\n\": exit status 1"
WARN[2020-08-17T20:42:47.335371850+01:00] could not use snapshotter zfs in metadata plugin  error="path /var/lib/docker/containerd/daemon/io.containerd.snapshotter.v1.zfs must be a zfs filesystem to be used with the zfs snapshotter"
INFO[2020-08-17T20:42:47.335706879+01:00] loading plugin "io.containerd.differ.v1.walking"...  type=io.containerd.differ.v1
INFO[2020-08-17T20:42:47.335835562+01:00] loading plugin "io.containerd.gc.v1.scheduler"...  type=io.containerd.gc.v1
INFO[2020-08-17T20:42:47.335994318+01:00] loading plugin "io.containerd.service.v1.containers-service"...  type=io.containerd.service.v1
INFO[2020-08-17T20:42:47.336143129+01:00] loading plugin "io.containerd.service.v1.content-service"...  type=io.containerd.service.v1
INFO[2020-08-17T20:42:47.336250164+01:00] loading plugin "io.containerd.service.v1.diff-service"...  type=io.containerd.service.v1
INFO[2020-08-17T20:42:47.336358125+01:00] loading plugin "io.containerd.service.v1.images-service"...  type=io.containerd.service.v1
INFO[2020-08-17T20:42:47.336474085+01:00] loading plugin "io.containerd.service.v1.leases-service"...  type=io.containerd.service.v1
INFO[2020-08-17T20:42:47.336584879+01:00] loading plugin "io.containerd.service.v1.namespaces-service"...  type=io.containerd.service.v1
INFO[2020-08-17T20:42:47.336694988+01:00] loading plugin "io.containerd.service.v1.snapshots-service"...  type=io.containerd.service.v1
INFO[2020-08-17T20:42:47.336798448+01:00] loading plugin "io.containerd.runtime.v1.linux"...  type=io.containerd.runtime.v1
INFO[2020-08-17T20:42:47.337097979+01:00] loading plugin "io.containerd.runtime.v2.task"...  type=io.containerd.runtime.v2
INFO[2020-08-17T20:42:47.337302067+01:00] loading plugin "io.containerd.monitor.v1.cgroups"...  type=io.containerd.monitor.v1
INFO[2020-08-17T20:42:47.339161952+01:00] loading plugin "io.containerd.service.v1.tasks-service"...  type=io.containerd.service.v1
INFO[2020-08-17T20:42:47.339371689+01:00] loading plugin "io.containerd.internal.v1.restart"...  type=io.containerd.internal.v1
INFO[2020-08-17T20:42:47.339581517+01:00] loading plugin "io.containerd.grpc.v1.containers"...  type=io.containerd.grpc.v1
INFO[2020-08-17T20:42:47.339721088+01:00] loading plugin "io.containerd.grpc.v1.content"...  type=io.containerd.grpc.v1
INFO[2020-08-17T20:42:47.339830697+01:00] loading plugin "io.containerd.grpc.v1.diff"...  type=io.containerd.grpc.v1
INFO[2020-08-17T20:42:47.339935547+01:00] loading plugin "io.containerd.grpc.v1.events"...  type=io.containerd.grpc.v1
INFO[2020-08-17T20:42:47.340039156+01:00] loading plugin "io.containerd.grpc.v1.healthcheck"...  type=io.containerd.grpc.v1
INFO[2020-08-17T20:42:47.340143487+01:00] loading plugin "io.containerd.grpc.v1.images"...  type=io.containerd.grpc.v1
INFO[2020-08-17T20:42:47.340248855+01:00] loading plugin "io.containerd.grpc.v1.leases"...  type=io.containerd.grpc.v1
INFO[2020-08-17T20:42:47.340353538+01:00] loading plugin "io.containerd.grpc.v1.namespaces"...  type=io.containerd.grpc.v1
INFO[2020-08-17T20:42:47.340459628+01:00] loading plugin "io.containerd.internal.v1.opt"...  type=io.containerd.internal.v1
INFO[2020-08-17T20:42:47.340651309+01:00] loading plugin "io.containerd.grpc.v1.snapshots"...  type=io.containerd.grpc.v1
INFO[2020-08-17T20:42:47.340762455+01:00] loading plugin "io.containerd.grpc.v1.tasks"...  type=io.containerd.grpc.v1
INFO[2020-08-17T20:42:47.340885730+01:00] loading plugin "io.containerd.grpc.v1.version"...  type=io.containerd.grpc.v1
INFO[2020-08-17T20:42:47.341042486+01:00] loading plugin "io.containerd.grpc.v1.introspection"...  type=io.containerd.grpc.v1
INFO[2020-08-17T20:42:47.341650676+01:00] serving...                                    address=/var/run/docker/containerd/containerd-debug.sock
INFO[2020-08-17T20:42:47.341891782+01:00] serving...                                    address=/var/run/docker/containerd/containerd.sock
INFO[2020-08-17T20:42:47.342010761+01:00] containerd successfully booted in 0.017493s
INFO[2020-08-17T20:42:47.351829436+01:00] pickfirstBalancer: HandleSubConnStateChange: 0x40007c6850, READY  module=grpc
INFO[2020-08-17T20:42:47.362524333+01:00] stopping healthcheck following graceful shutdown  module=libcontainerd
Error starting daemon: rename /var/lib/docker/runtimes /var/lib/docker/runtimes-old: invalid cross-device link
```

TerryE, 2023-02-01:
@rigon, this isn't an issue with `raspi-config` but rather a limitation of how docker itself uses overlayFS and interacts with overlayFS on the calling host.  I use overlayFS for my docker host RPis, but bind `/var/docker` to a separate persistent  LVM volume.

rigon, 2023-02-01:
You're right. I ended up adopting the same solution, but with a regular volume not LVM.

closed 2023-02-01 by rigon