RPi-Distro/raspi-config #133: Disable Bluetooth and Wifi?

ralphrmartin, 2020-09-18:
It would be handy if there were configuration options in raspi-config to disable Bluetooth and Wifi, for users who do not need them, or indeed must not transmit RF.

andrum99, 2020-09-19:
Add the following to /boot/config.txt:

`dtoverlay=disable-wifi
dtoverlay=disable-bt`

ralphrmartin, 2020-09-19:
Yes, I realise you can do that, but I am suggesting it would help other people if these were options in raspi-config.
 It seems at least as useful as some of the other things in there.

XECDesign, 2020-09-21:
Can't say I have any strong feelings on this one.

If you're in a situation where you must not transmit RF, whatever you're doing is outside of the scope of what raspi-config is for. The UI already offers these options through rfkill instead. It would be a little strange to have raspi-config and the UI do the same thing in different ways.

@spl237 ?

spl237, 2020-09-21:
Agree with @XECDesign - there are already GUI options to do this via rfkill; adding a second way of doing it is only going to cause confusion.

ralphrmartin, 2020-09-21:
"If you're in a situation where you must not transmit RF, whatever you're doing is outside of the scope of what raspi-config is for. " So living near a radio telescope, for example, makes you out of scope?

rfkill works after the system is up and running. While rfkill.service sets radio state "early in the boot process", I see no guarantee that the radio wont be on for a short time during boot.

Much other hardware is setup / enabled / disabled in raspi-config, so this would seem a logical place to expect to find it, and indeed where it should be to set switches that control how the system boots.

Anyway, I have made my suggestion, and said my piece. It seems like this simple addition to raspi-config is meeting with resistance. So be it.

andrum99, 2020-09-21:
> "If you're in a situation where you must not transmit RF, whatever you're doing is outside of the scope of what raspi-config is for. " So living near a radio telescope, for example, makes you out of [...]

I believe what was meant was that raspi-config is not meant to be used in the case where you must guarantee that there is no RF transmitters enabled on the board. An option to insert the relevant lines in config.txt to disable the onboard wifi and bluetooth would certainly be useful.

closed 2020-11-27 by XECDesign