RPi-Distro/raspi-config #134: Is non-interactive mode of raspi-config working?

JukkaKeisala, 2020-09-22:
Just installed `2020-08-20-raspios-buster-armhf-lite` on Raspberry Pi 3B+. Interactive mode of `raspi-config` is working correctly, but commanding `sudo raspi-config noint ...` hops to interactive mode.

I have read that non-interactive mode should work, is this the case?

Non-interactive mode would be needed for automated tasks.

XECDesign, 2020-09-23:
It's not really documented, since it was just meant as a way to keep rc_gui and raspi-config ways of doing things consistent and avoid having to make changes in multiple places. You can see how rc_gui uses it here - https://github.com/raspberrypi-ui/rc_gui/blob/master/src/rc_gui.c#L50

JukkaKeisala, 2020-09-24:
When I write, for example, `sudo raspi-config noint get_hostname` the interactive menu starts. No string-response is returned. That is the problem I have. For that command I would expect `raspberrypi` as return response.
I am doing it right?

XECDesign, 2020-09-24:
That should be `nonint` rather than `noint`

JukkaKeisala, 2020-09-24:
> That should be `nonint` rather than `noint`

That was it. I need to check my eyes :)

closed 2020-09-24 by JukkaKeisala